zmkfirmware/zmk #2182: Unable to pair halves

deividaspetraitis, 2024-02-29:
Hello,

I am trying to pair two Corne ( nice nano v2 ) halves but it seems I can not get it working. This is what I have already done trying to resolve the issue:

1. Reset both controllers based on the instructions provided here: https://zmk.dev/docs/troubleshooting
2. Install different firmwares

Any suggestions would be helpful!

Logs for the central side:

```
[17:00:06.864] Connected
n gpio@50000300 for output
[00:00:00.341,491] <dbg> zmk: kscan_matrix_init_output_inst: Configured pin 2 on gpio@50000000 for output
[00:00:00.341,552] <dbg> zmk: kscan_matrix_init_output_inst: Configured pin 29 on gpio@50000000 for output
[00:00:00.341,583] <dbg> zmk: kscan_matrix_init_output_inst: Configured pin 31 on gpio@50000000 for output
[00:00:00.341,644] <dbg> zmk: zmk_mouse_ps2_init: Inside zmk_mouse_ps2_init
[00:00:00.341,674] <dbg> zmk: zmk_mouse_ps2_init: Creating mouse_ps2 init thread.
[00:00:00.341,735] <dbg> zmk: vddh_init: VDDHDIV5 setup returned 0
*** Booting Zephyr OS build bd2a6c4862e0 ***
[00:00:00.343,383] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.343,414] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:00.343,444] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 3.5 Build 0
[00:00:00.343,811] <inf> bt_hci_core: No ID address. App must call settings_load()
[00:00:00.343,933] <dbg> zmk: ble_profiles_handle_set: Setting BLE value peripheral_addresses/0
[00:00:00.346,008] <inf> bt_hci_core: Identity: F1:E4:7C:0A:05:37 (random)
[00:00:00.346,038] <inf> bt_hci_core: HCI: version 5.4 (0x0d) revision 0x0000, manufacturer 0x05f1
[00:00:00.346,069] <inf> bt_hci_core: LMP: version 5.4 (0x0d) subver 0xffff
[00:00:00.349,243] <dbg> zmk: zmk_ble_ready: ready? 0
[00:00:00.349,243] <dbg> zmk: update_advertising: advertising from 0 to 2
[00:00:00.350,280] <dbg> zmk: start_scanning: Scanning successfully started
[00:00:00.351,501] <wrn> usb_hid: Device data not found for cfg 0x20001de8
[00:00:00.351,531] <dbg> zmk: zmk_usb_get_conn_state: state: 2
[00:00:00.351,562] <dbg> zmk: zmk_usb_get_conn_state: state: 2
[00:00:00.351,562] <dbg> zmk: get_selected_transport: No endpoint transports are ready.
[00:00:00.351,593] <dbg> zmk: zmk_hid_mouse_clear: Mouse report cleared
[00:00:00.351,593] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[00:00:00.351,623] <wrn> usb_hid: Device is not configured
[00:00:00.351,623] <err> zmk: FAILED TO SEND OVER USB: -11
[00:00:00.351,654] <dbg> zmk: zmk_endpoints_send_report: usage page 0x0C
[00:00:00.351,654] <wrn> usb_hid: Device is not configured
[00:00:00.351,684] <err> zmk: FAILED TO SEND OVER USB: -11
[00:00:00.351,715] <inf> zmk: Endpoint changed: BLE:0
[00:00:00.351,837] <dbg> zmk: zmk_usb_get_conn_state: state: 2
[00:00:00.351,867] <dbg> zmk: get_selected_transport: No endpoint transports are ready.
[00:00:00.351,928] <inf> zmk: Welcome to ZMK!

[00:00:00.354,614] <inf> usb_hid: Device suspended
[00:00:00.354,675] <dbg> zmk: zmk_usb_get_conn_state: state: 5
[00:00:00.354,675] <dbg> zmk: zmk_usb_get_conn_state: state: 5
[00:00:00.354,705] <dbg> zmk: get_selected_transport: Only USB is ready.
[00:00:00.354,705] <dbg> zmk: zmk_hid_mouse_clear: Mouse report cleared
[00:00:00.354,736] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[00:00:00.354,766] <dbg> zmk: zmk_endpoints_send_report: usage page 0x0C
[00:00:00.354,827] <inf> zmk: Endpoint changed: USB
[00:00:00.354,888] <dbg> zmk: destination_connection: Address pointer 0x2000ab3b
[00:00:00.354,888] <wrn> zmk: Not sending, no active address for current profile
[00:00:00.354,919] <dbg> zmk: destination_connection: Address pointer 0x2000ab3b
[00:00:00.354,919] <wrn> zmk: Not sending, no active address for current profile
[00:00:00.356,842] <dbg> zmk: vddh_sample_fetch: ADC raw 3005 ~ 4400 mV => 100%
[00:00:00.356,842] <dbg> zmk: zmk_battery_update: Setting BAS GATT battery level to 100.
[00:00:00.477,783] <dbg> zmk: split_central_device_found: [DEVICE]: A4:C1:38:8C:73:0F (public), AD evt type 0, AD data len 21, RSSI -77
[00:00:00.477,813] <dbg> zmk: split_central_eir_parse: [AD]: 1 data_len 1
[00:00:00.477,813] <dbg> zmk: split_central_eir_parse: [AD]: 22 data_len 16
[00:00:00.482,452] <dbg> zmk: split_central_device_found: [DEVICE]: 7C:4C:B4:21:2C:89 (random), AD evt type 2, AD data len 28, RSSI -56
[00:00:00.595,458] <dbg> zmk: split_central_device_found: [DEVICE]: 00:7C:2D:E3:C4:A5 (public), AD evt type 0, AD data len 31, RSSI -78
[00:00:00.595,458] <dbg> zmk: split_central_eir_parse: [AD]: 1 data_len 1
[00:00:00.595,489] <dbg> zmk: split_central_eir_parse: [AD]: 255 data_len 26
[00:00:00.606,994] <wrn> ps2_uart: UART RX detected error for byte 0xaa: Framing error (4)
[... 172 lines cut ...]
[00:00:00.337,219] <inf> bt_hci_core: No ID address. App must call settings_load()
[00:00:00.338,836] <inf> bt_hci_core: Identity: E8:AA:6F:FB:E6:5A (random)
[00:00:00.338,867] <inf> bt_hci_core: HCI: version 5.4 (0x0d) revision 0x0000, manufacturer 0x05f1
[00:00:00.338,897] <inf> bt_hci_core: LMP: version 5.4 (0x0d) subver 0xffff
[00:00:00.343,170] <inf> zmk: Welcome to ZMK!

[00:00:00.343,994] <dbg> zmk: zmk_usb_get_conn_state: state: 2
[00:00:00.347,137] <dbg> zmk: zmk_usb_get_conn_state: state: 5
[00:00:00.348,083] <dbg> zmk: vddh_sample_fetch: ADC raw 3046 ~ 4460 mV => 100%
[00:00:00.348,114] <dbg> zmk: zmk_battery_update: Setting BAS GATT battery level to 100.
[00:00:00.512,481] <dbg> zmk: zmk_usb_get_conn_state: state: 1
[00:00:00.577,697] <dbg> zmk: zmk_usb_get_conn_state: state: 1
[00:00:00.664,062] <dbg> zmk: zmk_usb_get_conn_state: state: 3
[00:00:30.592,132] <dbg> zmk: kscan_matrix_read: Sending event at 1,4 state on
[00:00:30.592,254] <dbg> zmk: zmk_kscan_process_msgq: Row: 1, col: 4, position: 16, pressed: true
[00:00:30.592,285] <dbg> zmk: split_listener:
[00:00:30.592,407] <dbg> zmk: send_position_state_callback: Error notifying -128
[00:00:30.593,170] <dbg> zmk: vddh_sample_fetch: ADC raw 3045 ~ 4460 mV => 100%
[00:00:30.689,147] <dbg> zmk: kscan_matrix_read: Sending event at 1,4 state off
[00:00:30.689,270] <dbg> zmk: zmk_kscan_process_msgq: Row: 1, col: 4, position: 16, pressed: false
[00:00:30.689,300] <dbg> zmk: split_listener:
[00:00:30.689,392] <dbg> zmk: send_position_state_callback: Error notifying -128
[00:00:30.787,139] <dbg> zmk: kscan_matrix_read: Sending event at 1,4 state on
[00:00:30.787,261] <dbg> zmk: zmk_kscan_process_msgq: Row: 1, col: 4, position: 16, pressed: true
```

caksoylar, 2024-02-29:
Instructions in https://zmk.dev/docs/troubleshooting#reset-split-keyboard-procedure are the only recommendation we have[^1]. I'd make sure to follow the instructions exactly, and try a few more times if it doesn't work.

(Instead of bootloader mode instructions, you can turn the controllers off/on if you have power switches. The important part is that they don't run the installed firmware during those times.)

[^1]: There is only one caveat, if you used e.g. CircuitPython on the controllers the flash doesn't work properly with ZMK; you might need a special firmware that clears the whole flash section.

deividaspetraitis, 2024-03-01:
Hey @caksoylar thanks for your input, as per your suggestion I have went to reset firmware using [zmk_clear.uf2](https://github.com/joric/nrfmicro/wiki/Bootloader) and afterwards I was able to pair successfully both halves. Marking as resolved, thank you!

closed 2024-03-01 by deividaspetraitis