zmkfirmware/zmk #2183: Documentation: more warning for Kconfig.shield

01110, 2024-02-29:
Hello!

I've just created a new keyboard shield based on the documentation: https://zmk.dev/docs/development/new-shield

I have written an extra space character between "shields_list_contains," and "my_board":

```
config SHIELD_MY_BOARD
    def_bool $(shields_list_contains, my_board)
```

The valid format is without space:

```
config SHIELD_MY_BOARD
    def_bool $(shields_list_contains,my_board)
```

The caused error was that the board does not advertised itself over bluetooth. After 2 hours of debugging and reading solved issues in the discord server's history I managed to fix the issue. The documentation does warn about that "my_board" string should be set properly, but as a person not familiar with writing Kconfig files I went over it at least 5 times (and the error is hard to find visually).

I think other people will mess this up in the future, so I suggest to add a warning or a sentence in the documentation that there should be absolutely no space where I added.

caksoylar, 2024-02-29:
It would be worth adding a warning box to that section, given I personally gave this advice tens of times over Discord when debugging issues.

huber-th, 2024-03-03:
@caksoylar I took a stab at this update to the documentation. Please let me know if you think the wording or the location of the warning box should be changed.

PR #2185 (2024-03-05): feat(docs): Add warning to new shield docs how Kconfig treats whitspaces
Resolves #2183

Added a warning to the shield section explaining that Kconfig does not ignore whitespaces on function calls and therefore adding whitespaces after the comma will break functionality.
closed 2024-03-05 by caksoylar