zmkfirmware/zmk #2184: Encoder on peripheral side stopped working

p-avital, 2024-03-02:
Hi there,

I've just built a Sofle v2 with Nice!Nanos and have been tinkering my ZMK config since. I got the encoders working on both sides at first, but I can't seem to get the one on the peripheral side to send keycodes anymore.

I did check that the solders were good, so I think this may be a configuration issue, I've made my config public [here](https://github.com/p-avital/zmk-config) in case you want to take a look.

I enabled USB debugging to try and check that the rotation was indeed detected, and it appears to be (I spun the encoder one click on the main board first, then spun the one on the peripheral board once).
```
[02:22:41.391,204] <dbg> zmk: set_wpm_symbol: WPM changed to 0
[02:22:45.481,689] <dbg> zmk: zmk_keymap_sensor_event: layer: 5 sensor_index: 0, binding name: enc_key_press
[02:22:45.481,750] <dbg> zmk: zmk_behavior_sensor_rotate_common_accept_data: val1: -2, val2: -500000, remainder: -2/-500000 triggers: 0 inc keycode 0xC00E9 dec keycode 0xC00EA
[02:22:45.481,781] <dbg> zmk: zmk_keymap_sensor_event: layer: 4 sensor_index: 0, binding name: enc_key_press
[02:22:45.481,811] <dbg> zmk: zmk_behavior_sensor_rotate_common_accept_data: val1: -2, val2: -500000, remainder: -2/-500000 triggers: 0 inc keycode 0x70051 dec keycode 0x70052
[02:22:45.481,872] <dbg> zmk: zmk_keymap_sensor_event: layer: 3 sensor_index: 0, binding name: enc_key_press
[02:22:45.481,903] <dbg> zmk: zmk_behavior_sensor_rotate_common_accept_data: val1: -2, val2: -500000, remainder: -2/-500000 triggers: 0 inc keycode 0xC00EA dec keycode 0xC00E9
[02:22:45.481,964] <dbg> zmk: zmk_keymap_sensor_event: layer: 2 sensor_index: 0, binding name: enc_key_press
[02:22:45.481,994] <dbg> zmk: zmk_behavior_sensor_rotate_common_accept_data: val1: -2, val2: -500000, remainder: -2/-500000 triggers: 0 inc keycode 0x107001D dec keycode 0x307001D
[02:22:45.482,055] <dbg> zmk: zmk_keymap_sensor_event: layer: 1 sensor_index: 0, binding name: enc_key_press
[02:22:45.482,086] <dbg> zmk: zmk_behavior_sensor_rotate_common_accept_data: val1: -2, val2: -500000, remainder: -2/-500000 triggers: 0 inc keycode 0xC00EA dec keycode 0xC00E9
[02:22:45.482,147] <dbg> zmk: zmk_keymap_sensor_event: layer: 0 sensor_index: 0, binding name: enc_key_press
[02:22:45.482,177] <dbg> zmk: zmk_behavior_sensor_rotate_common_accept_data: val1: -2, val2: -500000, remainder: -2/-500000 triggers: 0 inc keycode 0x70052 dec keycode 0x70051
[02:22:45.560,180] <dbg> zmk: zmk_keymap_sensor_event: layer: 5 sensor_index: 0, binding name: enc_key_press
[02:22:45.560,211] <dbg> zmk: zmk_behavior_sensor_rotate_common_accept_data: val1: -2, val2: -500000, remainder: 0/0 triggers: -1 inc keycode 0xC00E9 dec keycode 0xC00EA
[02:22:45.560,272] <dbg> zmk: zmk_keymap_sensor_event: layer: 4 sensor_index: 0, binding name: enc_key_press
[02:22:45.560,302] <dbg> zmk: zmk_behavior_sensor_rotate_common_accept_data: val1: -2, val2: -500000, remainder: 0/0 triggers: -1 inc keycode 0x70051 dec keycode 0x70052
[02:22:45.560,363] <dbg> zmk: zmk_keymap_sensor_event: layer: 3 sensor_index: 0, binding name: enc_key_press
[02:22:45.560,394] <dbg> zmk: zmk_behavior_sensor_rotate_common_accept_data: val1: -2, val2: -500000, remainder: 0/0 triggers: -1 inc keycode 0xC00EA dec keycode 0xC00E9
[02:22:45.560,424] <dbg> zmk: zmk_keymap_sensor_event: layer: 2 sensor_index: 0, binding name: enc_key_press
[02:22:45.560,455] <dbg> zmk: zmk_behavior_sensor_rotate_common_accept_data: val1: -2, val2: -500000, remainder: 0/0 triggers: -1 inc keycode 0x107001D dec keycode 0x307001D
[02:22:45.560,516] <dbg> zmk: zmk_keymap_sensor_event: layer: 1 sensor_index: 0, binding name: enc_key_press
[02:22:45.560,546] <dbg> zmk: zmk_behavior_sensor_rotate_common_accept_data: val1: -2, val2: -500000, remainder: 0/0 triggers: -1 inc keycode 0xC00EA dec keycode 0xC00E9
[02:22:45.560,607] <dbg> zmk: zmk_keymap_sensor_event: layer: 0 sensor_index: 0, binding name: enc_key_press
[02:22:45.560,638] <dbg> zmk: zmk_behavior_sensor_rotate_common_accept_data: val1: -2, val2: -500000, remainder: 0/0 triggers: -1 inc keycode 0x70052 dec keycode 0x70051
[02:22:45.560,699] <dbg> zmk: zmk_behavior_sensor_rotate_common_process: Sensor binding: enc_key_press
[02:22:45.560,729] <dbg> zmk: behavior_queue_process_next: Invoking key_press: 0x70051 0x00
[02:22:45.560,760] <dbg> zmk: on_keymap_binding_pressed: position 0 keycode 0x70051
[02:22:45.560,791] <dbg> zmk: hid_listener_keycode_pressed: usage_page 0x07 keycode 0x51 implicit_mods 0x00 explicit_mods 0x00
[02:22:45.560,791] <dbg> zmk: zmk_hid_implicit_modifiers_press: Modifiers set to 0x00
[02:22:45.560,821] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[02:22:45.560,882] <dbg> zmk: behavior_queue_process_next: Processing next queued behavior in 5ms
[02:22:45.560,913] <dbg> zmk: zmk_keymap_sensor_event: sensor event processing complete, behavior response was opaque
[02:22:45.565,979] <dbg> zmk: behavior_queue_process_next: Invoking key_press: 0x70051 0x00
[02:22:45.566,009] <dbg> zmk: on_keymap_binding_released: position 0 keycode 0x70051
[02:22:45.566,009] <dbg> zmk: wpm_event_listener: key_pressed_count 1 keycode 81
[02:22:45.566,040] <dbg> zmk: hid_listener_keycode_released: usage_page 0x07 keycode 0x51 implicit_mods 0x00 explicit_mods 0x00
[02:22:45.566,040] <dbg> zmk: zmk_hid_implicit_modifiers_release: Modifiers set to 0x00
[02:22:45.566,070] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[02:22:45.566,131] <dbg> zmk: behavior_queue_process_next: Processing next queued behavior in 0ms
[02:22:46.391,174] <dbg> zmk: wpm_work_handler: Raised WPM state changed 12 wpm_update_counter 1
[02:22:46.391,204] <dbg> zmk: set_wpm_symbol: WPM changed to 12
[02:22:46.583,892] <dbg> zmk: split_central_sensor_notify_func: [SENSOR NOTIFICATION] data 0x20010f73 length 12
[02:22:46.583,984] <dbg> zmk: peripheral_sensor_event_work_callback: Trigger sensor change for 1
[02:22:46.584,014] <dbg> zmk: zmk_keymap_sensor_event: layer: 5 sensor_index: 1, binding name: enc_key_press
[02:22:46.584,045] <dbg> zmk: zmk_behavior_sensor_rotate_common_accept_data: val1: 2, val2: 500000, remainder: 0/0 triggers: 0 inc keycode 0xA07001D dec keycode 0x807001D
[02:22:46.584,106] <dbg> zmk: zmk_keymap_sensor_event: layer: 4 sensor_index: 1, binding name: enc_key_press
[02:22:46.584,136] <dbg> zmk: zmk_behavior_sensor_rotate_common_accept_data: val1: 2, val2: 500000, remainder: 0/0 triggers: 0 inc keycode 0x7004F dec keycode 0x70050
[02:22:46.584,197] <dbg> zmk: zmk_keymap_sensor_event: layer: 3 sensor_index: 1, binding name: enc_key_press
[02:22:46.584,228] <dbg> zmk: zmk_behavior_sensor_rotate_common_accept_data: val1: 2, val2: 500000, remainder: 0/0 triggers: 0 inc keycode 0x70050 dec keycode 0x7004F
[02:22:46.584,289] <dbg> zmk: zmk_keymap_sensor_event: layer: 2 sensor_index: 1, binding name: enc_key_press
[02:22:46.584,320] <dbg> zmk: zmk_behavior_sensor_rotate_common_accept_data: val1: 2, val2: 500000, remainder: 0/0 triggers: 0 inc keycode 0x107001D dec keycode 0x307001D
[02:22:46.584,381] <dbg> zmk: zmk_keymap_sensor_event: layer: 1 sensor_index: 1, binding name: enc_key_press
[02:22:46.584,411] <dbg> zmk: zmk_behavior_sensor_rotate_common_accept_data: val1: 2, val2: 500000, remainder: 0/0 triggers: 0 inc keycode 0xC00B6 dec keycode 0xC00B5
[02:22:46.584,472] <dbg> zmk: zmk_keymap_sensor_event: layer: 0 sensor_index: 1, binding name: enc_key_press
[02:22:46.584,503] <dbg> zmk: zmk_behavior_sensor_rotate_common_accept_data: val1: 2, val2: 500000, remainder: 0/0 triggers: 0 inc keycode 0x70050 dec keycode 0x7004F
[02:22:47.391,174] <dbg> zmk: wpm_work_handler: Raised WPM state changed 6 wpm_update_counter 2
[02:22:47.391,204] <dbg> zmk: set_wpm_symbol: WPM changed to 6
[02:22:48.391,204] <dbg> zmk: wpm_work_handler: Raised WPM state changed 4 wpm_update_counter 3
[02:22:48.391,235] <dbg> zmk: set_wpm_symbol: WPM changed to 4
[02:22:49.391,174] <dbg> zmk: wpm_work_handler: Raised WPM state changed 3 wpm_update_counter 4
[02:22:49.391,204] <dbg> zmk: set_wpm_symbol: WPM changed to 3
[02:22:50.391,174] <dbg> zmk: wpm_work_handler: Raised WPM state changed 2 wpm_update_counter 5
[02:22:50.391,265] <dbg> zmk: set_wpm_symbol: WPM changed to 2
[02:22:51.391,174] <dbg> zmk: wpm_work_handler: Raised WPM state changed 0 wpm_update_counter 1
[02:22:51.391,235] <dbg> zmk: set_wpm_symbol: WPM changed to 0
```

Sorry for bothering you, I expect this is a case of PEBCAK, but I just can't get to work again (even by flashing previous iterations of the firmware, oddly enough...)

lookforrachel, 2024-05-10:
I just added an encoder to the peripheral half of my keyboard and it did not work but when I made the encoder side the central side, the encoder worked. I'm assuming it's because of the same issue you're having. I was initially guessing that it was because I didn't have encoders on both sides but seeing as you have encoders on both halves and are still experiencing this problem, that the encoder just doesn't work on peripheral side at all anymore

p-avital, 2024-05-10:
I have since resolved the issue: when soldering my keyboard's LEDs, I accidentally linked an LED's VCC with one of the traces for an alternative position for the encoder. 

This was one of the output trace, so one of the encoder's output would never change state, leading to no rotation being ever measured.

Cleaning up the solder fixed the issue completely.

closed 2024-05-10 by p-avital