zmkfirmware/zmk #2190: First press on mod-morph is not detected.

climent, 2024-03-07:
Hi.
I have some code using mod-morph to mod the ctrl-tab:

CTRL-TAB works under Chrome to move to the next Chrome tab. SHIFT-CTRL-TAB is used to move to the previous Chrome tab. I added a mod-morph so that when I press CTRL-TAB with the CMD modifier pressed, the behavior changes in a way that it behaves like pressing SHIFT-CTRL-TAB.

It used to work correctly but since the last update to head (Zephyr 3.5) the behavior has changed and now i have to repeat the TAB press 2 times, as the first CMD-CTRL-TAB does nothing. Only the second press actually produces the correct result. CTRL-TAB has continued working as expected.

climent, 2024-03-07:
Code is here: https://github.com/climent/zmk-config/blob/main/config/atreyu.keymap

climent, 2024-03-07:
I am closing this bug as it looks like it is a bad interaction with Karabiner-Elements in MacOS

closed 2024-03-07 by climent

zrkrlc, 2024-09-07:
Were you able to solve this anyhow?

climent, 2024-09-07:
I did not. I just stopped using Karabiner-Elements.