zmkfirmware/zmk #2192: [Feature Request] Separated Keymap Updating over UF2 DFU

myst729, 2024-03-08:
Given that ZMK supports UF2 firmware update, is it possible to just drag a keymap file (YAML, JSON or whatever, even plain TXT) under its DFU mode? For example:

```
<
&kp N1  &kp N2  &kp N3
&kp Q   &kp W   &kp E
&kp A   &kp S   &mo 1
>
<
&kp F1  &kp F2  &kp F3
&kp Z   &kp X   &kp C
&kp ESC &kp TAB &trans
>
```

For ZMK users who don't know coding at all and cannot setup a zmk-config repo on GitHub, this could really help. And on ZMK side, I guess this might be less cost than integrating VIAL or other solutions.

The downside is that users need to learn the concept of keycode and compose their own keymap file. However this can be improved in a more user friendly way, such as a single web page in the browser.

elpekenin, 2024-03-08:
In my opinion, setting up a fork of `zmk-config` is quite trivial, so im unsure how useful this would be to end users.
1. Still need to write your keymap using ZMK's syntax
2. If you need to jump into bootloader to drop the new bindings, might as well drop the UF2 compiled by GitHub Actions.

Anyway, as per the idea suggested... Copying a file onto the UF2 "fake" storage is not a moment/place where ZMK can run any logic, as it is part of the bootloader program.

Even in the scenario where the bootloader would parse such file, it would still be quite a lot of work (or even impossible) to get the contents of the file written into the datastructure(s) that ZMK uses for storing keymap/behavior information.

Also, there is "ZMK Studio" in the works, which would be a VIA-like tool (along other features?)

myst729, 2024-03-10:
> Also, there is "ZMK Studio" in the works, which would be a VIA-like tool (along other features?)

I know this project. It looks like in an "onhold" status. Any inside information or ETA? 🥹

joelspadin, 2024-03-10:
Pete is currently working on this. There's more discussion about it on the discord server, but https://github.com/zmkfirmware/zmk/issues/2186 is the latest update. No ETA yet, because it will be *a lot* of work.

nmunnich, 2025-08-15:
Closed by ZMK Studio.

closed 2025-08-15 by nmunnich