zmkfirmware/zmk #2215: Any membrane keyboards with ZMK support

korpe1, 2024-03-20:
Are there any membrane keyboards that ZMK supports?

caksoylar, 2024-03-20:
No, at least not that I know of. Membrane keyboards aren't really DIY territory, and among mass market manufacturers only ergonomic keyboard vendors like Kinesis and MoErgo seems to be using ZMK.

closed 2024-03-20 by caksoylar

korpe1, 2024-03-20:
> No, at least not that I know of. Membrane keyboards aren't really DIY territory, and among mass market manufacturers only ergonomic keyboard vendors like Kinesis and MoErgo seems to be using ZMK.

Can we add support for any membrane keyboard?

caksoylar, 2024-03-20:
Assuming they use a matrix or direct pin, sure. There is no anti-ghosting functionality in ZMK, and it might be necessary since they don't use diodes. #1533 and #1664 have been attempts at that.