zmkfirmware/zmk #2219: Action Build Error on default configuration

SparrowMike, 2024-03-22:
Hi!

Got my first corne with Nordic nRF52840 DK ( that's what I was told )

I have tried several times and always get the same result when building, waste quite a bit of time looking around and can't find anyone else having the same issue 

https://github.com/SparrowMike/zsk-config-corne/actions/runs/8393067110/job/22987218007

englmaxi, 2024-03-22:
Are you sure you use a [Nordic nRF52840 DK](https://www.nordicsemi.com/Products/Development-hardware/nRF52840-DK)?

The Corne is designed to be used with a [Pro Micro compatible board](https://zmk.dev/docs/hardware#pro_micro), which the Nordic DK is not. That's why your build failed.

SethMilliken, 2024-03-22:
You almost certainly have a `nice_nano_v2` clone not this: https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/zephyr/boards/arm/nrf52840dk_nrf52840/doc/index.html

SparrowMike, 2024-03-22:
Thanks @englmaxi and @SethMilliken yes I didn't think its the [Nordic nRF52840 DK](https://www.nordicsemi.com/Products/Development-hardware/nRF52840-DK) but that's what I was told to use when configuring 😅

Looking again at the messages with the seller he said the MCU was Nrf52840ble - however looking at the pictures it doesn't look the same as the top results from google.

[image]
[image]

SparrowMike, 2024-03-22:
I think this must be some sort of Supermini nRF52840 copy? hahaha honestly no clue

But it should be safe just to flush it with `nice_nano_2` right ?

englmaxi, 2024-03-22:
according to [this](https://github.com/joric/nrfmicro/wiki/Alternatives#52840nano), it uses the `nice_nano_v2` shield, yes.

SparrowMike, 2024-03-22:
I'm embarresed 😅

Thanks loads for your time, will give this a go. 

cheers!

closed 2024-03-22 by SparrowMike