zmkfirmware/zmk #2246: [Feature Request] Automatic Default Layer Switching Based on Bluetooth Profile in ZMK

ozkan, 2024-04-03:
**Description:**
As a user of ZMK firmware for custom keyboards, I often switch between different devices with varying Bluetooth profiles. Currently, switching between Bluetooth profiles manually is seamless with ZMK. However, it would greatly enhance the usability of ZMK firmware if it could automatically switch the default layer based on the selected Bluetooth profile.

**Proposal:**
Implement a feature in ZMK firmware that automatically switches the default layer based on the selected Bluetooth profile. This would allow users to have different key mappings tailored to each Bluetooth profile, eliminating the need for manual layer switching each time the Bluetooth profile changes.

**Benefits:**

- Improved user experience: Automatic default layer switching would streamline the keyboard setup process for users who frequently switch between devices.

- Increased versatility: Users would be able to customize key mappings for different Bluetooth profiles, enhancing the flexibility of ZMK firmware.

- Simplified workflow: Eliminating the need for manual layer switching saves time and effort for users.

elpekenin, 2024-04-03:
I plan to PR that soon, see #2222 for some info.

In the meantime, you can use (and give feedback, please) the intial implementation i have done  (https://github.com/elpekenin/zmk-userspace)

and-elf, 2024-06-11:
Wouldn't it easier to use a macro that chooses BT profile and selects a specific layer?

fitzage, 2024-12-13:
@and-elf Your comment was exactly the kick in the pants I needed to set up the macros. Perfect. No extra features required.