zmkfirmware/zmk #2252: Bluetooth connection lost when sending mod-morphs to M1 Mac

sallygal1, 2024-04-05:
My keyboard spontaneously loses connection when sending a mod-morph to my M1 Mac.

I'm using [this](https://github.com/urob/zmk-nodefree-config) library to send unicode mod-morphs to the Mac via bluetooth. Every so often when sending the mod-morph, the keyboard loses the bluetooth connection. My keyboard runs on batteries. I've enabled USB debugging (connection set to bluetooth) but the issue did not occur for 2 days (power via USB/battery charging on main half).

The only log entry I can see on the Mac is:

`Couldn't find active session 0x00000003289033H7! (status=65535)` 

After that I have to open the Mac's bluetooth menu, Forget the device, then I'm able to reconnect again. It'll continue working for a while, then it happens again. This only happens when the keyboard runs on the batteries. It seems to me there is either some weird race condition or some lock up of the chip (nice!nano v2.0) due to the mod morph.

I'm having some troubles sourcing the issues and I'm out of ideas what to try next.

caksoylar, 2024-04-05:
If this is a crash, it might be worth sharing a (ideally minimal) reproduces case

sallygal1, 2024-04-17:
The thing is, I can't reproduce it consistently enough (hence why I think it's a weird race-condition/lock-up). Is there a way to get the logs when running off battery? Like logging to a file on the chip, or sending logs via bluetooth.

caksoylar, 2024-04-17:
You can use USB logging but send keycodes to the active BLE profile: https://zmk.dev/docs/behaviors/outputs