zmkfirmware/zmk #2259: Macros do not respect &trans

Intuos5, 2024-04-07:
Currently, I cannot get a macro to respect &trans. The idea is that if a macro can leverage &trans, you could activate a key on a layer and follow it up with a different behavior (e.g. on key release). This could allow for accessing layers below the current layer, which can be followed up with a layer change, for instance.

d3m1gd, 2025-04-03:
same issue here. my use case is going to layer with several keys acting as arrow keys, and all the rest are mapped to &trans and go back to level 0. but both `&trans &to 0` and `&to 0 &trans` dont work, and simple idea of disabling layer and pressing underlying keys has to be implemented like the workaround mentioned above

caksoylar, 2025-04-03:
This is an architectural limitation, would likely be solved by #2848.