zmkfirmware/zmk #2260: [Feature request] Sequential combo 'flavor'

Intuos5, 2024-04-07:
By default, combos are evaluated based on whether a combination of keys is held within the timeout-ms threshold. This, however, means that the order in which the keys were pressed does not matter. I would like to request a sequential combo flavor, in which the order of the keystrokes also matters for a combo activation. This is interesting in combination with hold_taps and layer changing behaviors, because these require you to hold down a key. Pressing the next key then activates the combo, but not vice versa (e.g. key 1 followed by 2 does activate a combo, but 1 followed by 2 does not). 
With homerow mods, this sequential combo could also be used to prevent certain specific accidental chorded (rolled) capitalisations, the negation of which can than enable a further reduction of   `require-prior-idle-ms` and/ or the `tapping-term-ms`.

caksoylar, 2024-04-08:
For reference, QMK has a `COMBO_MUST_PRESS_IN_ORDER_PER_COMBO` option to do this: https://docs.qmk.fm/#/feature_combo?id=per-combo-timing-holding-tapping-and-key-press-order

AlejandroSanchez90, 2025-01-25:
Would love this feature!

Nick-Munnich, 2025-02-19:
I've been thinking about this a bit recently. I can't think of any use case where it would make sense to use a sequential combo rather than a positional hold-tap or a mod morph or something similar. I think those fill the role of "sequential combo" far better. That said, it could probably do with investigating making such a sequential hold-tap "combo" easier to configure.

Could someone interested in this PR perhaps provide a specific example where the above does not hold true?

badjeff, 2025-02-19:
Did combo "release order" has ever been discussed?
Let say the combo is trigger by {A+B}, we'd have extra logic for [release A then B], or [release B then A].

Nick-Munnich, 2025-02-19:
I don't recall any discussion around that, but that is a very interesting idea. I'll have to ponder that for a while.

Nick-Munnich, 2025-02-19:
> The way I could see this being used is if you have a two key combo, where activating A+B gives Ctrl+Tab and B+A Ctrl+Shift+Tab, basically making the combo directional. Now this is a simple example, but you get the point.

See for this case I think you could get it to work *now* with a combination of positional hold-tap, and either a mod-morph or an extra layer. Like I said, not particularly *easy* to do, and that part could probably use some improving, but conceptually I see this as *maybe* but not necessarily a combo

badjeff, 2025-02-19:
One use case would be adding a combo on top of 2 consecutive keys of `[` & `]`. Use new release orders for `{` & `}`.

Another use case would be added on timeless HRM, releasing the `Shift` at last to enable sticky-shift.

caksoylar, 2025-02-19:
> Did combo "release order" has ever been discussed? Let say the combo is trigger by {A+B}, we'd have extra logic for [release A then B], or [release B then A].

#1809 addresses that, I believe.

Intuos5, 2025-02-20:
Another use-case I have is that I use my split ergonomic keyboard with a pen tablet. In that case, I frequently only use the left half. While I could opt to add a layer where I disable positional hold tap, I would prefer to set up a timed system where if I use HRM Shift with another key on the left half, it either activates a combo (e.g. within a quick roll) or outside of the window, it is a Shift + Key. That way, I can use the same baselayer regardless of whether I am typing or using only one half of the keyboard with a pen tablet.

baggiiiie, 2025-03-17:
i think a use case would be similar to neovim config. e.g., `<leader>ab` to trigger some action.  
in this case, <leader> a b are pressed in a sequential order while within a timeout, enabling easier activation and more flexibility. 

> The way I could see this being used is if you have a two key combo, where activating A+B gives Ctrl+Tab and B+A Ctrl+Shift+Tab, basically making the combo directional. Now this is a simple example, but you get the point.

@Intuos5 's idea of this is actually pretty cool and useful, though im not sure about the difficult and necessity of implementation, as this could be probably done with other functionalities instead of combos.