zmkfirmware/zmk #2270: Error during build

spacemishka, 2024-04-13:
Hi,

I try to create a configuration for my dumbpad. But it gives my this error  and dont  know the reason. 

`
-- Found BOARD.dts: /tmp/zmk-config/zmk/app/boards/arm/nice_nano/nice_nano_v2.dts
-- Found devicetree overlay: /__w/dumbpad2/dumbpad2/boards/shields/dumbpad/dumbpad.overlay
-- Found devicetree overlay: /__w/dumbpad2/dumbpad2/boards/shields/dumbpad/dumbpad.keymap
devicetree error: /tmp/zmk-config/zephyr/misc/empty_file.c:1 (column 1): parse error: expected node name, property name, or '}'
CMake Error at /tmp/zmk-config/zephyr/cmake/modules/dts.cmake:279 (message):
-- In: /tmp/tmp.qoCtubDJiK/zephyr, command: /usr/bin/python3;/tmp/zmk-config/zephyr/scripts/dts/gen_defines.py;--dts;/tmp/tmp.qoCtubDJiK/zephyr/zephyr.dts.pre;--dtc-flags;'';--bindings-dirs;/tmp/zmk-config/zmk/app/module/dts/bindings;/tmp/zmk-config/zmk/app/dts/bindings;/tmp/zmk-config/zephyr/dts/bindings;--header-out;/tmp/tmp.qoCtubDJiK/zephyr/include/generated/devicetree_generated.h.new;--dts-out;/tmp/tmp.qoCtubDJiK/zephyr/zephyr.dts.new;--edt-pickle-out;/tmp/tmp.qoCtubDJiK/zephyr/edt.pickle;--vendor-prefixes;/tmp/zmk-config/zmk/app/dts/bindings/vendor-prefixes.txt;--vendor-prefixes;/tmp/zmk-config/zephyr/dts/bindings/vendor-prefixes.txt
  gen_defines.py failed with return code: 1
Call Stack (most recent call first):
  /tmp/zmk-config/zephyr/cmake/modules/zephyr_default.cmake:[12](https://github.com/spacemishka/dumbpad2/actions/runs/8675301540/job/23788598974#step:11:13)9 (include)
  /tmp/zmk-config/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /tmp/zmk-config/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:8 (find_package)
`
My configuration for my Gherkin runs fine. 

Any ideas?

caksoylar, 2024-04-13:
You are probably missing a closing brace `};` in your file, having a link to the failing build would make it much easier to figure out. Next time I'd encourage you to ask on [Discord](https://zmk.dev/community/discord/invite) where it is more appropriate to have a back and forth conversation on user issues.

spacemishka, 2024-04-13:
It is here. 
https://github.com/spacemishka/dumbpad2/tree/main

spacemishka, 2024-04-13:
Ok...there was missing bracket. Thanks for your help!

closed 2024-04-13 by spacemishka