zmkfirmware/zmk #2274: Soft off only turn off the left side of my Corne split keyboard

picrots, 2024-04-16:
I recently built a wireless Corne keyboard on the original v3 PCB design which doesn't have a power switch.

To turn my keyboard off I setup a keymap to trigger the Soft Off feature and it does work however only on the master side and the right side stays up.

I tried to create additional key map on the right side but that also only turn the left side off.

I am using Corne V3 and Nice!nano controller. My last build was through github action on 15th Apr 2024.

omalishe-pearl, 2024-04-23:
Same here, also there is a discord thread with some other people having same problem: https://discord.com/channels/719497620560543766/1225441220311126106/1225441220311126106

closed 2024-04-29 by petejohanson

picrots, 2024-05-10:
This problem has not been resolved for me yet and right side of my Corne stays awake.
Build date: 10 May 2024 (using github actions)

GermanG, 2024-05-10:
Can you check #2300? Also use my ZMK repo to verify if that's your case

PR #2285 (2024-04-29): shortening keymap_soft_off behavior label
This PR is aiming to fix #2274 
The behavior label gets truncated because the len is 9, defined in app/include/zmk/split/bluetooth/service.h as ZMK_SPLIT_RUN_BEHAVIOR_DEV_LEN
Shortening the keymap_soft_off solves the issue. 
Thanks bravekarma (discord) for the advice.