zmkfirmware/zmk #2279: Update development docs with new module instructions

infused-kim, 2024-04-20:
Hi,

I noticed a few missing instructions for the developer docs and wanted to raise them here.

## 1. zmk-config is a module now

I created a new zmk-config the other day and noticed that the format of it changed a bit. 

We discussed it in discord for a bit and I wanted to summerize the results here.

It's now a zephyr module. This means the `boards/` directory is in within `zmk-config/boards/` now. Previously it was in `zmk-config/config/boards/`.

This means that if you want to build a firmware with a custom shield, then you need to use `-DZMK_EXTRA_MODULES` in the command  like this now:

```bash
west build -d build/tc/right -p -b nice_nano_v2 -- -DZMK_EXTRA_MODULES="/workspaces/zmk-config/" -DZMK_CONFIG="/workspaces/zmk-config/config/" -DSHIELD="think_corney_right nice_view"
```

Notice that `DZMK_EXTRA_MODULES` module points to the `zmk-config/` directory and the `DZMK_CONFIG` to the config subdirectory `zmk-config/config/`.

If you want to include other extra modules, you have to separate them with a semicolon like this: `-DZMK_EXTRA_MODULES="/workspaces/zmk-config/;/workspaces/zmk-modules/kb_zmk_ps2_mouse_trackpoint_driver/"`

Which leads to the next thing...

## No VSCode docker volume point for `zmk-modules`

Currently, the VSCode container doesn't have a volume for zmk-modules.

In the meantime it can be achieved by creating the volume...

```bash
docker volume create --driver local -o o=bind -o type=none -o device="/full/path/to/your/zmk-modules/" zmk-modules
```

Next, edit in your zmk clone `zmk/.devcontainer/devcontainer.json`:

```diff
{
  "name": "ZMK Development",
  [...]
  "mounts": [
    "type=volume,source=zmk-root-user,target=/root",
    "type=volume,source=zmk-config,target=/workspaces/zmk-config",
+  "type=volume,source=zmk-modules,target=/workspaces/zmk-modules",
    "type=volume,source=zmk-zephyr,target=${containerWorkspaceFolder}/zephyr",
    "type=volume,source=zmk-zephyr-modules,target=${containerWorkspaceFolder}/modules",
    "type=volume,source=zmk-zephyr-tools,target=${containerWorkspaceFolder}/tools"
  ],
  [...]
}
```

Then, in VSCode press `Command + Shift + P` and execute the command `Dev Containers: Rebuild Without Cache and Reopen in Container`.

Once the container is rebuilt, you can delete the above line from your `devcontainer.json` so that it doesn't pollute your git history.

The container will continue mounting the `zmk-modules/` volume as long as you don't rebuild it.

caksoylar, 2024-04-20:
Re: point 1, it is documented in https://zmk.dev/docs/development/build-flash#building-with-external-modules. Maybe it can be made clearer that the config repo is a module as well and requires these instructions? If so, that could be done in the next section https://zmk.dev/docs/development/build-flash#building-from-zmk-config-folder with a pointer to the previous section, and perhaps modifying the example to add the module.

infused-kim, 2024-04-20:
> If so, that could be done in the next section https://zmk.dev/docs/development/build-flash#building-from-zmk-config-folder with a pointer to the previous section, and perhaps modifying the example to add the module.

Yeah I agree. I think that's the best place to put it.

caksoylar, 2024-07-25:
Closed by #2380

closed 2024-07-25 by caksoylar