zmkfirmware/zmk #2281: Keys used in combos are not reflected with mouse clicks before timeout.

joonhyuk, 2024-04-24:
I'm currently using both shift keys in a combo, and if I shift+click (select text from the current cursor) before the timeout, it just clicks.

I've investigated the keystrokes with the event viewer and found that for the keys used in the combo, the key down only happens after the timeout if I hold the key down. Of course, the collaboration with other keys or just tapping, it happens immediately without any problem.

As far as I can tell, the problem is only with mouse clicks.

caksoylar, 2024-04-24:
I think this is the expected behavior given the current feature set of combos. Essentially, when a key involved in a combo is pressed, it will wait for either:
- combo timeout to expire, or
- you press another key that is not in the combo, or release any key
to decide that it, in fact, doesn't resolve to a combo and should invoke the underlying behavior (shift here). Since the keyboard doesn't know about any mouse clicks, it cannot shortcut the decision like the second case.

In theory, the combos could have a property similar to the newly added [`hold-while-undecided`](https://zmk.dev/docs/behaviors/hold-tap#hold-while-undecided) property for hold-taps to invoke the mod before it resolves to the combo or not. But in practice, you can probably set the combo timeout short enough (like 60ms) that the delay is imperceptible for mouse use.

Dallas-J, 2024-07-09:
Just my two cents, but something similar to [hold-while-undecided](https://zmk.dev/docs/behaviors/hold-tap#hold-while-undecided) is probably pretty important to anyone who plays games. I was playing with a friend tonight and realized that tapping shift didn't register unless I held it for the full timeout. Even 60ms is pretty noticeable IMO, and lower values makes the combo hard to register.

@joonhyuk For shift specifically, you can use mod morphs to work around this behavior: https://github.com/Dallas-J/bleuet-shield/commit/1e0f734b6487f0c4947b71fc6d0e456cc171faed

oleasteo, 2025-08-04:
@caksoylar 
> Since the keyboard doesn't know about any mouse clicks, it cannot shortcut the decision like the second case.

Just to clarify: Are you talking about a separate mouse? I'm experiencing the same issue with mouse actions triggered by zmk itself. Those should be able to be fixed?