zmkfirmware/zmk #2288: caps word plus mods works only in some programms in Ubuntu

timLoewel, 2024-04-30:
I configured home row mods and caps word.

In my Ubuntu 22.04 on a Dell Laptop, in most programs (e.g.  gnome terminal, libre office, firefox) the letters, that have Mod-Tap or Layer-tap configured, are written in lower case during a caps word. In Visual Studio Code on the same system it works as expected.
Also on my Android phone it works.

e.g.
`asdfGZXcVB`
in Firefox.

`ASDFGZXCVB`
in VSCode and on my Android phone

Where asdf have mods configured, and c has a layer-tap.

Any idea?

Thanks, Tim

timLoewel, 2024-04-30:
I tested a bit with xev.

With caps word enabled the "a" (which only works in vscode) sends this:

```
KeyPress event, serial 28, synthetic NO, window 0x800001,
    root 0x4c2, subw 0x0, time 45297169, (66,75), root:(2011,591),
    state 0x0, keycode 50 (keysym 0xffe1, Shift_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyPress event, serial 28, synthetic NO, window 0x800001,
    root 0x4c2, subw 0x0, time 45297169, (66,75), root:(2011,591),
    state 0x1, keycode 38 (keysym 0x41, A), same_screen YES,
    XLookupString gives 1 bytes: (41) "A"
    XmbLookupString gives 1 bytes: (41) "A"
    XFilterEvent returns: False

KeyRelease event, serial 28, synthetic NO, window 0x800001,
    root 0x4c2, subw 0x0, time 45297170, (66,75), root:(2011,591),
    state 0x1, keycode 50 (keysym 0xffe5, Caps_Lock), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyRelease event, serial 28, synthetic NO, window 0x800001,
    root 0x4c2, subw 0x0, time 45297170, (66,75), root:(2011,591),
    state 0x0, keycode 38 (keysym 0x61, a), same_screen YES,
    XLookupString gives 1 bytes: (61) "a"
    XFilterEvent returns: False
```

And a "g" (which works as expected)

```
KeyPress event, serial 28, synthetic NO, window 0x800001,
    root 0x4c2, subw 0x800002, time 45413180, (48,53), root:(1993,569),
    state 0x0, keycode 50 (keysym 0xffe1, Shift_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyPress event, serial 28, synthetic NO, window 0x800001,
    root 0x4c2, subw 0x800002, time 45413180, (48,53), root:(1993,569),
    state 0x1, keycode 42 (keysym 0x47, G), same_screen YES,
    XLookupString gives 1 bytes: (47) "G"
    XmbLookupString gives 1 bytes: (47) "G"
    XFilterEvent returns: False

KeyRelease event, serial 28, synthetic NO, window 0x800001,
    root 0x4c2, subw 0x800002, time 45413272, (48,53), root:(1993,569),
    state 0x1, keycode 50 (keysym 0xffe5, Caps_Lock), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyRelease event, serial 28, synthetic NO, window 0x800001,
    root 0x4c2, subw 0x800002, time 45413272, (48,53), root:(1993,569),
    state 0x0, keycode 42 (keysym 0x67, g), same_screen YES,
    XLookupString gives 1 bytes: (67) "g"
    XFilterEvent returns: False

```
 
I see no difference.

And the output of `evtest --grab` when triggering capsword and then `agbc`.

```
Event: time 1714473515.007322, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e1
Event: time 1714473515.007322, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
Event: time 1714473515.007322, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70004
Event: time 1714473515.007322, type 1 (EV_KEY), code 30 (KEY_A), value 1
Event: time 1714473515.007322, -------------- SYN_REPORT ------------
Event: time 1714473515.007378, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e1
Event: time 1714473515.007378, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 0
Event: time 1714473515.007378, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70004
Event: time 1714473515.007378, type 1 (EV_KEY), code 30 (KEY_A), value 0
Event: time 1714473515.007378, -------------- SYN_REPORT ------------
Event: time 1714473515.585871, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e1
Event: time 1714473515.585871, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
Event: time 1714473515.585871, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7000a
Event: time 1714473515.585871, type 1 (EV_KEY), code 34 (KEY_G), value 1
Event: time 1714473515.585871, -------------- SYN_REPORT ------------
Event: time 1714473515.750879, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e1
Event: time 1714473515.750879, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 0
Event: time 1714473515.750879, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7000a
Event: time 1714473515.750879, type 1 (EV_KEY), code 34 (KEY_G), value 0
Event: time 1714473515.750879, -------------- SYN_REPORT ------------
Event: time 1714473522.560922, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e1
Event: time 1714473522.560922, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
Event: time 1714473522.560922, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70005
Event: time 1714473522.560922, type 1 (EV_KEY), code 48 (KEY_B), value 1
Event: time 1714473522.560922, -------------- SYN_REPORT ------------
Event: time 1714473522.725958, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e1
Event: time 1714473522.725958, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 0
Event: time 1714473522.725958, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70005
Event: time 1714473522.725958, type 1 (EV_KEY), code 48 (KEY_B), value 0
Event: time 1714473522.725958, -------------- SYN_REPORT ------------
Event: time 1714473523.460820, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e1
Event: time 1714473523.460820, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 1
Event: time 1714473523.460820, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70006
Event: time 1714473523.460820, type 1 (EV_KEY), code 46 (KEY_C), value 1
Event: time 1714473523.460820, -------------- SYN_REPORT ------------
Event: time 1714473523.461796, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e1
Event: time 1714473523.461796, type 1 (EV_KEY), code 42 (KEY_LEFTSHIFT), value 0
Event: time 1714473523.461796, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70006
Event: time 1714473523.461796, type 1 (EV_KEY), code 46 (KEY_C), value 0
Event: time 1714473523.461796, -------------- SYN_REPORT ------------
```

I also see no difference :frowning:

timLoewel, 2024-04-30:
I guess this is https://github.com/zmkfirmware/zmk/issues/759#issuecomment-1193182843 the cause?

Strangely it only occurs on tap keys.

timLoewel, 2024-04-30:
`pkill ibus`  from here https://github.com/zmkfirmware/zmk/issues/759#issuecomment-1433417201
worked. So it is the same issue.

closed 2024-04-30 by timLoewel

schmjop, 2024-05-08:
@timLoewel I have the same issue as you and `pkill ibus` solves it for me as well.

Were you able to automate the killing of `ibus`?
I added the line to my `~/.profile` and to my crontab with `@reboot pkill ibus` but after a reboot when checking if `ibus` is running with `pgrep ibus`, `ibus` is still running.
Thanks.