zmkfirmware/zmk #2293: Sticky layer documentation is unclear

shunlog, 2024-05-03:
It is not clear what's meant by "key press" in the sentence "A sticky layer stays pressed until another key is pressed." on the [sticky layer page](https://zmk.dev/docs/behaviors/sticky-layer). One would think it means when you physically press *any* key, but actually the sticky layer is only deactivated when a keycode is sent, so for example it isn't deactivated on a layer toggle, [or on caps word activation](https://github.com/zmkfirmware/zmk/issues/2133#issuecomment-1910838961). 

It should be clarified, either by replacing "key pressed" with "keycode sent", or adding an info box or something.

shunlog, 2024-05-03:
BTW, can someone explain why would it be desirable for the sticky layer to be disabled only on a keycode sent, instead of on any key-press?

caksoylar, 2024-05-03:
A docs update would be welcome. I also had this general thought on sticky keys:
> I think we should fix sticky keys/layers pages to be more like hold-tap and mod-taps: There should be an intro to the generic/user-defined `behavior-sticky-key` in the former, and it should also mention the pre-defined `&sk`. Sticky layer page should have a note box saying it is just a sticky key with `bindings = <&mo>`.

> BTW, can someone explain why would it be desirable for the sticky layer to be disabled only on a keycode sent, instead of on any key-press?

This is related to above, since sticky layer is simply an instance of sticky key, and sticky keys only listen to keycode change events (rather than "position" change events in ZMK parlance).

nguyendown, 2024-08-04:
'Press binding' is often used in the code base. The sticky key documentation probably avoids introducing too many concepts or terms to users. The term 'binding' becomes even less relevant since sticky key rarely requires advanced configuration, unlike Combo or Sensor. We should have a dedicated page to explain 'what is binding'. Then sticky key and other behaviors can be described much more easily and accurately.

> can someone explain why would it be desirable for the sticky layer to be disabled only on a keycode sent, instead of on any key-press?

It's been an ongoing issue.
https://github.com/zmkfirmware/zmk/issues/1689