zmkfirmware/zmk #2294: Getting constant battery indicator notifications on Linux.

Ay1tsMe, 2024-05-04:
I keep getting battery level notifications every few minutes on Linux when I don't use the keyboard after a while.

I'm not sure if there is a config option that turns off notifications for the keyboard. I'm pretty sure what's happening is it's sending a notification of the battery when the keyboard wakes up from sleeping. It would be nice if I could turn this off but still get a notification when it connects from a disconnect with the battery level as well.

ReFil, 2024-05-04:
You can turn off battery level monitoring with CONFIG_BT_BAS=n in your _defconfig or .conf file

Ay1tsMe, 2024-05-04:
> You can turn off battery level monitoring with CONFIG_BT_BAS=n in your _defconfig or .conf file

Will this turn off all monitoring of my battery or just the notifications. Because I want the notification to still appear but only when it connects to my PC from a disconnect?

joelspadin, 2024-05-04:
With that disabled, you will not be able to see the battery level at all.

This sounds like an issue with the OS more than the keyboard. Could you provide more information about your setup? "Linux" doesn't tell us much...

Ay1tsMe, 2024-05-05:
Using hyprland on EndevourOS. Im using mako to handle notifications. 

I dont have an issue with any of my other Bluetooth devices renotifying me about the battery level. I have a mx master 3s and it doesnt ping the battery.

JeffDess, 2024-05-07:
I am experiencing the same, although I never figured out it was triggered by a sleep/wake event. NiceNano + Arch Hyprland here, with blueman and swaync. I also use the reporting feature to monitor battery levels at al times in Waybar. Never had any issues with other devices either.

igorgue, 2024-08-01:
I use `dunst` on `Hyprland` on `NixOS` and this worked for me:

Edit: `.config/dunst/dunstrc`:

```toml
[ignore_kb_bat_notifications]
   appname = "blueman"
   summary = "Skree Custom"
   body = "Connected *%"
   skip_display = true
```

You can check your notification summary, appname, and other data (likely the message is different for other languages) with `dunstctl history`.

I think this will specifically help you @JeffDess

JeffDess, 2024-08-01:
Thanks @igorgue, that's neat. I believe there is a similar functionality in swaync:

```json
"notification-visibility": {
  "Bluetooth":{
    "app-name": "blueman",
    "state": "ignored",
  }
},
```

However, I am using Sway most of the time these days (Hyprland is way too buggy since this wlroots thing) and I cannot say I have experienced the issue for a while. 

Since everybody who commented here is using Hyprland and the timeframe also matches perfectly with the wlroots work, I'd say there a good chance that it was introduced by this new implementation (April~May).