zmkfirmware/zmk #2300: Soft off only turn off the right side of Corne split keyboard issues

GermanG, 2024-05-09:
This a sequel of #2274, which is partially solved in  #2285
The behavior is not consistent, most of the times works perfectly, but sometimes the right side is still on.
Reporting this now, I'll update my firmware to debug and update this issue when it happens.

GermanG, 2024-05-09:
On the right side there isn't any event, on the left I saw this
```
[00:00:07.966,278] <dbg> zmk: position_state_changed_listener: 29 bubble (no undeci
ded hold_tap active)                                                               
  
[00:00:07.966,308] <dbg> zmk: zmk_keymap_apply_position_state: layer: 2 position: 2
9, binding name: soft_off  
[00:00:07.966,369] <dbg> zmk: split_bt_invoke_behavior_payload:                    
[00:00:07.966,430] <dbg> zmk: split_central_split_run_callback:                                                                                                      
[00:00:13.844,177] <dbg> zmk: kscan_matrix_read: Sending event at 2,5 state
```

But when it works I saw:
```
[00:00:10.537,170] <dbg> zmk: kscan_matrix_read: Sending event at 2,5 state off
[00:00:10.537,261] <dbg> zmk: zmk_ksc
```

My educated guess is that the left side goes down before completing the request to the right side.

GermanG, 2024-05-09:
I added this to app/src/behaviors/behavior_soft_off.c
```
           if (!IS_SPLIT_PERIPHERAL) {
                k_usleep(100000);
            }
```
the whole fragment looks like
```
        if (hold_time > config->hold_time_ms) {
            if (!IS_SPLIT_PERIPHERAL) {
                k_usleep(100000);
            }
            zmk_pm_soft_off();
        } else {
            LOG_INF("Not triggering soft off: held for %d and hold time is %d", hold_time,
                    config->hold_time_ms);
        }
```

So far so good. I think the issue is introduced because of the event is triggered when the key is released, so it's a matter of luck.

petejohanson, 2024-05-10:
> I added this to app/src/behaviors/behavior_soft_off.c
> 
> ```
>            if (!IS_SPLIT_PERIPHERAL) {
>                 k_usleep(100000);
>             }
> ```
> 
> the whole fragment looks like
> 
> ```
>         if (hold_time > config->hold_time_ms) {
>             if (!IS_SPLIT_PERIPHERAL) {
>                 k_usleep(100000);
>             }
>             zmk_pm_soft_off();
>         } else {
>             LOG_INF("Not triggering soft off: held for %d and hold time is %d", hold_time,
>                     config->hold_time_ms);
>         }
> ```
> 
> So far so good. I think the issue is introduced because the event is triggered when the key is released, so it's a matter of luck.

I think this should be tweaked a bit to only add the sleep if we are specifically a split central. Does it work as expected with that in place?

GermanG, 2024-05-10:
So far my hack is working, I'm giving it more time to be sure. I'll open a PR if it goes ok for a couple of days.

closed 2024-05-27 by GermanG