zmkfirmware/zmk #2304: Xo

anuvyklack, 2024-05-12:


caksoylar, 2024-05-12:
The syntax for boolean properties is `slow-release;`, see https://zmk.dev/docs/config#bool-1 for a reference.

caksoylar, 2024-05-13:
Closing as user error, comment again if there is trouble.

closed 2024-05-13 by caksoylar

alfiedotwtf, 2024-10-02:
I just bumped into this too - I think the problem is the datatype is bool and the docs say the default value is "false". Maybe the docs should be changed to this being a switch - either it exists or not.

caksoylar, 2024-10-02:
I understand the confusion, would be nice to reduce it somehow. The above linked section does document it, and given [Zephyr seems to call these booleans as well](https://docs.zephyrproject.org/3.5.0/build/dts/bindings-syntax.html#type), I wouldn't want to deviate from that in ZMK docs.