zmkfirmware/zmk #2307: `hold-trigger-key-positions` does not seem to work with `flavor = "balanced"`

inoc603, 2024-05-20:
I'm trying to use `hold-trigger-key-positions` for positional homerow mods on a glove80 and [here](https://github.com/inoc603/zmk-config/blob/main/config/glove80.keymap) is my config.

I defined the homerow mods with https://github.com/urob/zmk-helpers like this:

```
#define HRM_POSITIONAL(tapping_term_ms, keys) \
	flavor = "balanced"; \
	tapping-term-ms = <tapping_term_ms>; \
	quick-tap-ms = <75>; \
	require-prior-idle-ms = <150>; \
	bindings = <&kp>, <&kp>; \
	hold-trigger-key-positions = <keys>; \
	hold-trigger-on-release;

ZMK_HOLD_TAP(hml, HRM_POSITIONAL(250, KEYS_R KEYS_T))
ZMK_HOLD_TAP(hmr, HRM_POSITIONAL(250, KEYS_L KEYS_T))
ZMK_HOLD_TAP(hmls, HRM_POSITIONAL(175, KEYS_R KEYS_T))
ZMK_HOLD_TAP(hmrs, HRM_POSITIONAL(175, KEYS_L KEYS_T))
```

The homerow mods works except that `hold-trigger-key-positions` does not seem to provide the corss-hand only behavior that I wanted and the homerow mods applies for all keys on both hands. I've tried with different bidings and without macro but ended up with the same result.

I tried with `flavor = "tap-unless-interrupted"` and it seems to work. I found a similar isssue https://github.com/zmkfirmware/zmk/issues/2040 but for another flavor. From the discussion there I'd assume my configuration should work with `balanced`.

My setup is paritially working but there are frequent miss fires. I'm kind of stuck now. Would appreciate if someone can take a look at this or give me some guidance on how to debug the problem.

caksoylar, 2024-05-20:
This is very hard to debug given all the helpers, can you share the simplified version? As for the premise, there haven't been other reports of this not working (and it seems to be working for me). You might be observing the effect of `tap-unless-interrupted` instead of `hold-trigger-key-positions` when you try that.

urob, 2024-05-21:
Are you sure they are triggering within the tapping term? If not, that's how the positional hold tap option is expected to work. 

Might be worth increasing your tapping terms to 2s (2000ms) or so to make sure.

inoc603, 2024-05-22:
> Are you sure they are triggering within the tapping term? If not, that's how the positional hold tap option is expected to work.
> 
> Might be worth increasing your tapping terms to 2s (2000ms) or so to make sure.

Guess I missed this section in the docs ....

```
If the LEFT_SHIFT / Q key is held by itself for longer than tapping-term-ms, a hold behavior is produced. This is because positional hold-tap only modifies the behavior of a hold-tap if another key is pressed before the tapping-term-ms period expires.
```

Thanks for the clarification! the behavior should be expected. I'll try tweaking the tapping term to reduce miss fires

closed 2024-05-22 by inoc603