zmkfirmware/zmk #2311: Cannot get nice!view to work after recent update

ghostbuster91, 2024-05-26:
Hi, 

I have a custom keyboard with a self made shield - https://github.com/ghostbuster91/Dilemma3x6_3-zmk
This is a split keyboard equipped with nice!nanos & nice!views. More about the keyboard [here](https://github.com/ghostbuster91/Dilemma3x6_3).

Today I updated the firmware in the left half (master). Previous firmware version was from almost a year ago.
I am using a fork of zmk which is fairly up-to-date: https://github.com/nickconway/zmk/

The keyboard used to work well so far with both displays.

The update didn't went smoothly as first the build failed with following errors:
```
/opt/zephyr-sdk-0.16.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/libzephyr.a(status.c.obj):/__w/Dilemma3x6_3-zmk/Dilemma3x6_3-zmk/zmk/app/boards/shields/nice_view/widgets/status.c:284: multiple definition of `widget_layer_status_mutex'; app/libapp.a(layer_status.c.obj):/__w/Dilemma3x6_3-zmk/Dilemma3x6_3-zmk/zmk/app/src/display/widgets/layer_status.c:51: first defined here
/opt/zephyr-sdk-0.16.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/libzephyr.a(status.c.obj):/__w/Dilemma3x6_3-zmk/Dilemma3x6_3-zmk/zmk/app/boards/shields/nice_view/widgets/status.c:284: multiple definition of `widget_layer_status_work'; app/libapp.a(layer_status.c.obj):/__w/Dilemma3x6_3-zmk/Dilemma3x6_3-zmk/zmk/app/src/display/widgets/layer_status.c:51: first defined here
```

At first, I decided to just comment out anything custom connected with the display and see if the build will pass ([commit](https://github.com/ghostbuster91/Dilemma3x6_3-zmk/pull/3/commits/ef668f0a6cca2a4a142bdc4a127d602c8c4b5c40)). 

It did so I installed new firmware on the board. After that, I noticed that the display stopped working. It started to show some gibberish. I thought that maybe this is due to some misconfiguration on my side. 

I found [this issue](https://github.com/zmkfirmware/zmk/issues/1273#issuecomment-1712858158) which mentions the same errors and applied the suggestion. This fixed the compilation so I uploaded new firmware on the board. However, the screen remained borked. 

After this I decided to build the old firmware once again using the zmk revision from the day the of the last commit in my repository. That was 19 Jul, so I took f3110d1d1ea195725551b4c997c542a4003e1452 which is right before:
  
[image]

The build passed without any issues - https://github.com/ghostbuster91/Dilemma3x6_3-zmk/pull/4 and I uploaded it on the board. However the screen issue remains. 

Below, photo of both halves, the right one with updated firmware and the left one with the original one. 
[image]

1. Do you think that the display might have gotten irreversibly broken? 
2. If so what could have caused that?

caksoylar, 2024-05-26:
I doubt your display is broken by flashing different firmwares. Just in case the setting got scrambled, I would try toggling external power off/on so that you can make sure it is on on both sides. Then wait a minute (so the setting is persisted to flash), and reset the controller. (Sometimes display doesn't boot up properly on ext. power on but it can boot after a reset.)

It looks like you figured out what happened: The nice!view got a custom status screen in the last year, and the .conf settings you used don't apply to them.

If I were you I wouldn't bother testing the old firmware. I'd flash the new one (if you want you can disable the custom screen https://github.com/zmkfirmware/zmk/issues/1273#issuecomment-1712938706 and keep your .conf settings) and test that. As an aside, you might want to use ZMK main along with https://github.com/dhruvinsh/zmk-tri-state module instead of nickconway's ZMK fork (which needs to be manually updated).

ghostbuster91, 2024-06-02:
> Then wait a minute (so the setting is persisted to flash)

What setting do you have in mind here? 

I tried restarting it on a battery and on ext_power. Still the issue persists :(

caksoylar, 2024-06-03:
> What setting do you have in mind here?

I mean the internally stored state of the external power toggle. When you use e.g. `&ext_power EP_ON`, the state gets saved to storage after one minute (to reduce flash wear on repeated uses).

I don't have other suggestions, but if you make a help post on the Discord maybe other folks can comment.

ghostbuster91, 2024-06-23:
Got you. Thanks for your suggestions. I will look for more help on discord. 

I am going to close the issue for now. If I get some new information that will be worth sharing I will post it here and we might then reopen.

MalpenZibo, 2025-03-28:
Ok, in my case I just discovered that the issue was a bad solder of the cs nice view PIN.

So no firmware issue in my case.

nmunnich, 2025-08-04:
I'm going to re-close this, considering the current state of the issue. Feel free to re-open if you see a reason to do so, but I'm happy to write this off as a hardware error.

closed 2025-08-04 by nmunnich

ghostbuster91, 2025-11-08:
> It would be a good idea to create a branch in your repository with these proposed changes in order to test them out.

I did that before writing in here. It works.
https://github.com/ghostbuster91/Dilemma3x6_3-zmk/blob/nix2/build.yaml#L15

> I would suggest [overriding the CS pin in your keymap](https://github.com/search?q=path%3A%2F%5Econfig%5C%2F.%2B%5C.keymap%24%2F+%26nice_view_spi+cs-gpios&type=code).

While this solution would probably work I am not a fan of it because I think that such configuration belongs more to the shield definition than to the keymap.