zmkfirmware/zmk #2316: Inconsistent behavior when using &caps_word

swisscows, 2024-05-30:
Hi all, 

I am getting strange behavior when using the &caps_word behavior behind a combo. I will get seemingly random lowercase characters thrown into my string of uppercase chars.

[image]

All of the above instances were obtained by hitting my combo for caps_word, typing h e l l o and then hitting space and reactivating caps_word.

The relevant parts of my config are below.

Board:   Corne
Shield:   nice!nano v2
OS:         Fedora 39 and Windows 10

```c
&caps_word {
    continue-list = <UNDERSCORE MINUS DOT COMMA BSPC DEL LSHIFT RSHIFT LCTRL RCTRL LALT RALT LMETA RMETA>;
};

#define BASE  0
#define COMBO_TERM_VERY_SLOW 60
#define COMBO_PRIOR_IDLE 100

 #define COMBO(NAME, BINDINGS, KEYPOS, LAYERS, MS) \
    combo_##NAME { \
        timeout-ms = <MS>; \
        require-prior-idle-ms = <COMBO_PRIOR_IDLE>; \
        key-positions = <KEYPOS>; \
        layers = <LAYERS>; \
        bindings = <BINDINGS>; \
    };

COMBO(caps,          &caps_word, 31 34, BASE, COMBO_TERM_VERY_SLOW)
```

As a list of things I have ruled out.
- The software I am typing in, this issue occurs across multiple different unrelated software and even on different machines.
- The board itself, I have reseated the microcontrollers and the issue seems random as it will happen across random keys and I don't have any issues with any keys when caps_word is not active.
- Anything somehow deactivating caps_word by mistake as I will continue to get capitalized characters after getting lowercase ones. I can even type, then backspace a char repeatedly and sometimes it will be uppercase and sometimes lowercase while caps_word is active.

If anyone has any insight about what may be going on I would appreciate the help. Thank you.

caksoylar, 2024-05-30:
Sounds like it is related to #759. Fedora 39 should have Gnome 45 which in theory doesn't have this issue because it has the fixed Mutter. It might still be worth checking the version.

swisscows, 2024-05-30:
Interesting, I am actually on the KDE spin using kwin version 5.27.11. When I connect to windows 10 I connect directly, not via RDP.

swisscows, 2024-05-30:
Actually, after looking at #2288 the ibus package seems to be at fault here. Killing it resolves the issue and restarting it brings it back.

swisscows, 2024-05-30:
Closing, as "not a zmk issue".

closed 2024-05-30 by swisscows