zmkfirmware/zmk #2325: Sticky key behaviour ignores `release-after-ms` property

jradtilbrook, 2024-06-06:
I wanted to reduce the timeout of all sticky keys to 300ms as seen in [this config](https://github.com/jradtilbrook/zmk-config/blob/fdb53ffc095e6f5d9865c8aa00007d143ed3c209/config/settings.dtsi#L19-L27). But when I flash that to my keyboard, I can see that it still has the 1s timeout.

I've verified this using a stopwatch, starting when I tap "shift" and pressing it some time after 300ms. After multiple attempts at this at different timings, it appears that the 1s timeout is what prevails.

I'm happy to contribute a PR if this is a bug and someone can point me in the right direction. Or show me how my config is wrong (FWIW I tested out setting `lazy` and that did work which leads me to believe the way I'm configuring it is working but this property is not)

caksoylar, 2024-06-06:
It doesn't look like your settings made it to the final devicetree: https://github.com/jradtilbrook/zmk-config/actions/runs/9401996069/job/25895035480#step:13:739

I am guessing that this https://github.com/jradtilbrook/zmk-config/blob/fdb53ffc095e6f5d9865c8aa00007d143ed3c209/config/sofle.keymap#L15 coming after the settings include might have reset the values to their defaults.

jradtilbrook, 2024-06-06:
Ohh thank you @caksoylar! That makes some sense. I suppose since `lazy` has to be "deleted" as well, that would be the reason why that is still working 👍 

I'll try that when I'm back at that keyboard

jradtilbrook, 2024-06-07:
Yes that has fixed it. Thank you!!