zmkfirmware/zmk #2329: Problem with mod-morph and volume media keys

Ciebiada, 2024-06-10:
Hi Guys!

I want to bind `LALT + G` to `C_VOL_UP`

I have this, which works with everything else, but not in this case
```c
behaviors {
        vol_up: volume_up {
                compatible = "zmk,behavior-mod-morph";
                #binding-cells = <0>;
                bindings = <&kp G>, <&kp C_VOL_UP>;
                mods = <(MOD_LALT)>;
        }
}
``` 

Instead of increasing volume my Mac opens volume settings 🤯. When I bind `C_VOL_UP` to a key on a layer it works just fine.

Am I missing something?

caksoylar, 2024-06-10:
Do you have Karabiner Elements installed?
[image]

Ciebiada, 2024-06-11:
I've had it installed in the past, but uninstalled some time ago. Do you think it could have a permanent effect, even after it is uninstalled and not running.
Also I have a bunch of other mod-morphs defined e.g. LGUI + f -> right arrow. Which works fine, sending just right arrow.

caksoylar, 2024-06-11:
It shouldn't, no. What key combination is supposed to open the volume settings? (I don't own a Mac.)

Ciebiada, 2024-06-11:
Apparently it’s option + volume. Which makes sense. So it looks like the code I posted is still sending LGUI mod for some reason

caksoylar, 2024-06-11:
This might be similar issue: https://github.com/zmkfirmware/zmk/issues/1685#issuecomment-1791931449

I suspect the mod masking of mod-morphs don't work with keycodes from the consumer usage page like `C_VOL_UP`.