zmkfirmware/zmk #2339: Feature request: Hyper, MEH and other combination modifier shortcuts

nicenemo, 2024-06-18:
I would like to add a few defines for combinations of modifiers, such as 
* HYPER (Shift, Control, Alt and Gui)
* MEH, the same but without GUI
* Combinations of two modifiers

For all combinations an all left and an all right should exist, e.g. `LHYP` and `RHYP`

For all combinations a variant with a key should exist, e.g. `LHYPK(key)` and `RHYPK(key)`

Proper documentation should be provided.
 
This allows:
* for easy starting of office programs on Windows. Microsoft redefined 'Hyper' as the 'OfficeKey' and it allows you to start office programs out of the box, e.g. `Hyper X` for Excel and ` Hyper O` for Outlook. `Hyper L` starts the LinkedIn webpage. When office is not installed the web version will be started.
* Easy defenition of the copilot key which is `Left Shift Gui F23` as `LSG(F23)`
* Easy definition of Function key layers with modifiers that do not interfer with any programs
* Easy definition of shortcuts for shells or terminals.  In Powershell  you cannot define shortcuts with a GUI key.
* Just have modtap combination  modifiers available below the home row as many programs already use combination modifiers.

caksoylar, 2024-07-10:
In order to understand the scope of changes that should go into something like #2341, I think it would be helpful to elaborate a bit more on this issue. Specifically, I am curious on how you think a new approach with specific defines like suggested would improve upon the status quo of combining individual modifier functions+modifier keycodes to achieve the same function. As far as I can tell, the issue definition doesn't address this specifically.

This is documented in https://zmk.dev/docs/codes/modifiers#modifier-functions, with examples showing combinations of modifiers. Do you think improving the docs on the existing approach would help? For example, we could add pointers on defining Hyper modifier and modifier functions that people could use in their keymaps.

urob, 2024-07-11:
I don't have an opinion on MEH and HYPER. 

But I see merit in adding a copilot key. The main argument here is that users don't need to know what it maps to in order to productively use it. (This is different to HYPER etc, which still require knowledge of what modifiers they map to in order to define shortcuts using them on the OS side).

Nick-Munnich, 2024-07-11:
`HYPER` commonly gets used outside of office for user defined shortcuts, to avoid conflicts due to the number of modifiers. I agree that a dedicated `COPILOT` key could be valuable. I don't see any value in defining modifier combinations for 2 modifiers, and am neutral towards `MEH` but don't see the value in other triple-modifier combinations.

nicenemo, 2024-07-11:
I Use MEH for some shortcuts because Microsoft claimed hyper as the office key.

I have a copilot alias of in my personal zmk-config repo

Nick-Munnich, 2024-08-02:
Thinking on this some more, I think it would make more sense for the defines to be either a part of or something akin to [urob's zmk helpers](https://github.com/urob/zmk-helpers). I'm not sold on having this directly in ZMK as it's mostly a convenience thing, but it would work nicely as an optional convenience add-on. 

It might be nice to document certain keyboard shortcuts such as `copilot` or unicode input in the ZMK docs somewhere without any source changes, though. "OS Software Aliases", perhaps.

caksoylar, 2024-08-02:
The copilot key combination can be documented as an example under modifier functions.

nguyendown, 2024-08-03:
Users can easily define C macros in their keymap. This feature is unnecessary.

I don't think we should use any copyrighted/brand names in the document either.

nicenemo, 2024-08-03:
I do understand the argument against Copilot. That would also fly for the Windows key.

Regarding these defines, QMK has them. QMK even has two modifier combinations like I had in my original proposal. 

Yes users can define a lot in their keymap. That is how this started. The keyboard setup I use, uses the same defines from my keymap, not from this proposal. 

As stated earlier, I only think for keyboard editors having such defines by default would make it a bit easier. I am happy to use my defines in my keymap. I lf anybody wants to take this further go ahead.

Nick-Munnich, 2024-08-03:
I would suggest taking your earlier PR and turning the content into a [ZMK
module](https://zmk.dev/docs/features/modules), moving the docs changes
into the readme. Then you could also add the 2 and 3 modifier combinations
back. Once we have a centralised place to collect modules, it could then be
listed under there so that interested users can find it.

What would be your thoughts on that solution?

nmunnich, 2025-03-29:
We won't be adding the definitions themselves to ZMK, but there is definitely a need to document these definitions. I've opened #2899 to track that need separately.