zmkfirmware/zmk #2354: Cannot turn off Bluetooth with config (NRF52840 supermini)

rorymcd98, 2024-07-02:
I have this [board](https://github.com/joric/nrfmicro/wiki/Alternatives#supermini-nrf52840),  It mimics a Nice!Nano so I have selected the Nano while building my firmware. 

I am just using the board in wired mode (I don't need bluetooth / wireless). My concern is that anyone can connect to my keyboard at any time.

This is my whole config:

```
# Uncomment the following lines to enable the Corne RGB Underglow
# CONFIG_ZMK_RGB_UNDERGLOW=y
# CONFIG_WS2812_STRIP=y

# Uncomment the following line to enable the Corne OLED Display
# CONFIG_ZMK_DISPLAY=y
CONFIG_BT=n
#Uncommenting the below was giving me issues
#CONFIG_BT_MAX_CONN=0
```

Expect: My keyboard not to be connectable to via bluetooth.

Current: My keyboard is on bluetooth and any device can connect to it.

This is my whole repo: https://github.com/rorymcd98/zmk-config

rorymcd98, 2024-07-04:
@lesshonor thanks, this didn't really work because it disabled pairing alltogether (even when I bumped to 2 connections). But this is not a major concern for me any more. 

Feel free to close he issue.

closed 2024-07-04 by petejohanson

ldab, 2025-04-24:
it seems there is a dependency, this works for me:

```
CONFIG_ZMK_BLE=n
CONFIG_BT=n
```