zmkfirmware/zmk #2355: Build failure for out-of-the-box Sofle +

phinate, 2024-07-02:
Apologies in advance if I miss out any information — complete ZMK first-timer.

I've just recieved a [pre-built Sofle v1 purchased from here](https://mechboards.co.uk/products/sofle-wireless-hotswap-kit-pre-build), which has NRF52840 Bluetooth Controllers. After selecting Sofle (option 70) and NRF52840 M.2 (option 17) in the respective first two steps, I have a `devicetree error` in my GHA build:
```
devicetree error: /tmp/zmk-config/zmk/app/boards/shields/sofle/sofle.dtsi:71 (column 1): parse error: undefined node label 'pro_micro_i2c'
```

You can see the build [here](https://github.com/phinate/zmk-config/actions/runs/9763652939/job/26950064289#step:11:48) for context.

Any help debugging what's going on would be fantastic!

caksoylar, 2024-07-02:
From the link in https://zmk.dev/docs/hardware, this is the `nrf52840_m2` board you selected: https://wiki.makerdiary.com/nrf52840-m2/

If they ship with supermini controllers, then you should use `nice_nano_v2`. The vendor should have had some instructions for this, ideally.

closed 2024-07-02 by caksoylar

phinate, 2024-07-02:
Perfect. Thanks for allowing for my ignorance here, and sorry for not reaching out to the vendor first!