zmkfirmware/zmk #2359: sticky layer on another layers

Hoffmano, 2024-07-03:
My desired setup

Layer 0 - conditional layer to layer 4
layer 4 - &to layer 5
layer 5 - &sl layer 2

but on layer 5 I can't do sticky layer or even momentary layer, only to layer works

any tip ou trick about this

setup explanation
layer 0 is my daily layer
layer 2 is my number layer
layer 5 is my gaming layer, without any modifiers on homing row
laayer 4 is a layer hub

Hoffmano, 2024-07-03:
I've made a test using sticky layer on default layer and work properly

Hoffmano, 2024-07-03:
my setup
https://github.com/Hoffmano/zmk-config

caksoylar, 2024-07-03:
FYI you can edit your original post, no need to post extra comments.

When you use `&sk 2` from layer 5 (that is turned on with `&to 5`) layer 5 is still active. This means that when you press a key, layer 5 will capture it before layer 2, unless it has `&trans`. This priority order is described in https://zmk.dev/docs/features/keymaps#layers. So reorder your layers so that you aren't activating a lower-indexed layer from a higher-indexed layer.