zmkfirmware/zmk #2361: Display not initialising after recent PR

NightB1ade, 2024-07-04:
#2358 appears to have had an unintended consequence that the OLED displays (on my board at least) are not showing anything any more.

Tested:
- Waterfowl R2 board with OLEDs
- nine!nano v2
- Clean test using a new repo with clean install, only activating display in config, then targeting revisions in `west.yml`
    - Using `f18974e8c401b139058549dcd95ab7a4cdc7a6e9` revision: Build is normal and display works
    - Using `80173f8ea33e592bba2aa59ae08f036ec44f4e81` revision: Build does not, display shows nothing
- Note: The rest of the keyboard (keymap etc.) appears to work fine - just the OLED display

Repo here: https://github.com/NightB1ade/waterfowl-zmk - will be iterating my keymap later on the repo - so specific commits are:
- `9c8b5280428c6be2e9ec057749b12f45d7c47db0` - display working
- `4dd4cab7fc882b6cb1c12eeeb6e40b02229e7d24` - display not working

petejohanson, 2024-07-04:
@NightB1ade can you please gather logging of the startup sequence?

My *guess* is that the OLED init sequence may be occuring before the VCC power is enabled, but I want to be sure. Also: does VCC properly read 3.3v on the new firmware?

NightB1ade, 2024-07-04:
[2024_07_04_1743 Waterfowl Logs.txt](https://github.com/user-attachments/files/16101345/2024_07_04_1743.Waterfowl.Logs.txt)

Does this cover the right bit?  I scanned it and can see some relevant entries, but don't know enough to know if it's all you need!

petejohanson, 2024-07-04:
@NightB1ade Please give #2362 a try by following the instructions on https://zmk.dev/docs/features/beta-testing and let me know if that fixes things for you. TIA!

NightB1ade, 2024-07-04:
It worked!  Another new repo, repointed the west.yml (see below to check I got it right), and then activated the display only in config.  Thank you!

```
manifest:
  remotes:
    - name: zmkfirmware
      url-base: https://github.com/zmkfirmware
    - name: petejohanson
      url-base: https://github.com/petejohanson
    # Additional modules containing boards/shields/custom code can be listed here as well
    # See https://docs.zephyrproject.org/3.2.0/develop/west/manifest.html#projects
  projects:
    - name: zmk
      remote: petejohanson
      revision: core/oled-init-delayed-settings-load-fix
      import: app/west.yml
  self:
    path: config
```

PR #2362 (2024-07-04): fix: Fix external power settings load issue.
* Because settings load is delayed now, enable external
  power on init, and let it be disabled on settings load
  later, to ensure power is on early for
  other peripheral initialization.

Fixes: #2361
commit e01f13f9 (2024-07-04): fix: Fix external power settings load issue.

* Because settings load is delayed now, enable external
  power on init, and let it be disabled on settings load
  later, to ensure power is on early for
  other peripheral initialization.

Fixes: #2361
closed 2024-07-04 by petejohanson