zmkfirmware/zmk #2372: Enabling USB_LOGGING and reading the ttyACM device makes the keyboard fail

JuantAldea, 2024-07-14:
Context: Lotus58  + nice!nano compatible board. Left hand only.

Upon opening the debug tty device, nothing will be printed, but the keyboard will stop sending keystrokes. After a few key-presses , the following messages will be printed upon each keystroke:

```
➜  ~ tio /dev/ttyACM0                                                     
[11:29:47.762] tio v3.5
[11:29:47.762] Press ctrl-t q to quit
[11:29:51.207] Connected to /dev/ttyACM0
:25:50.995,666] <wrn> usb_device: Failed to write endpoint buffer 0x83
[00:25:50.995,697] <wrn> usb_device: Failed to write endpoint buffer 0x83
[00:25:50.995,727] <wrn> usb_device: Failed to write endpoint buffer 0x83
[00:25:50.995,758] <err> zmk: FAILED TO SEND OVER USB: -11
[00:25:51.077,178] <dbg> zmk: kscan_matrix_read: Sending event at 2,3 state on
[00:25:51.077,270] <dbg> zmk: zmk_physical_layouts_kscan_process_msgq: Row: 2, col: 3, position: 27, pressed: true
[00:25:51.077,301] <dbg> zmk: position_state_changed_listener: 27 bubble (no undecided hold_tap active)
[00:25:51.077,362] <dbg> zmk: zmk_keymap_apply_position_state: layer: 0 position: 27, binding name: key_press
[00:25:51.077,392] <dbg> zmk: on_keymap_binding_pressed: position 27 keycode 0x70007
[00:25:51.077,423] <dbg> zmk: hid_listener_keycode_pressed: usage_page 0x07 keycode 0x07 implicit_mods 0x00 explicit_mods 0x00
[00:25:51.077,423] <dbg> zmk: zmk_hid_implicit_modifiers_press: Modifiers set to 0x00
[00:25:51.077,453] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[00:25:51.093,170] <dbg> zmk: kscan_matrix_read: Sending event at 2,4 state off
[00:25:51.093,261] <dbg> zmk: zmk_physical_layouts_kscan_process_msgq: Row: 2, col: 4, position: 28, pressed: false
[00:25:51.093,292] <dbg> zmk: position_state_changed_listener: 28 bubble (no undecided hold_tap active)
[00:25:51.093,353] <dbg> zmk: zmk_keymap_apply_position_state: layer: 0 position: 28, binding name: f_right
[00:25:51.093,383] <dbg> zmk: on_keymap_binding_released: position 28 keycode 0x70009
[00:25:51.093,414] <dbg> zmk: hid_listener_keycode_released: usage_page 0x07 keycode 0x09 implicit_mods 0x00 explicit_mods 0x00
[00:25:51.093,444] <dbg> zmk: zmk_hid_implicit_modifiers_release: Modifiers set to 0x00
[00:25:51.093,444] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[00:25:51.123,596] <wrn> usb_device: Failed to write endpoint buffer 0x83
[00:25:51.123,626] <wrn> usb_device: Failed to write endpoint buffer 0x83
[00:25:51.123,657] <wrn> usb_device: Failed to write endpoint buffer 0x83
[00:25:51.123,687] <wrn> usb_device: Failed to write endpoint buffer 0x83
[00:25:51.123,718] <wrn> usb_device: Failed to write endpoint buffer 0x83
[00:25:51.123,748] <wrn> usb_device: Failed to write endpoint buffer 0x83
[00:25:51.123,748] <wrn> usb_device: Failed to write endpoint buffer 0x83
[00:25:51.123,779] <wrn> usb_device: Failed to write endpoint buffer 0x83
[00:25:51.123,809] <wrn> usb_device: Failed to write endpoint buffer 0x83
[00:25:51.123,840] <wrn> usb_device: Failed to write endpoint buffer 0x83
[00:25:51.123,870] <wrn> usb_device: Failed to write endpoint buffer 0x83
[00:25:51.123,901] <err> zmk: FAILED TO SEND OVER USB: -11
[00:25:51.123,901] <dbg> zmk: zmk_hid_masked_modifiers_clear: Modifiers set to 0x00
[00:25:51.209,167] <dbg> zmk: kscan_matrix_read: Sending event at 2,3 state off
[...]
``` 
So to me it looks like reading from the ttyACM conflicts with the USB HID. Is this normal?

caksoylar, 2024-07-14:
I don't think this is normal, it isn't intended and I haven't seen any other reports of this.

JuantAldea, 2024-07-26:
This seems to be caused by my own computer.  It works fine if I connect it through a USB HUB, or to a different machine.

closed 2024-07-26 by JuantAldea