zmkfirmware/zmk #2375: Macros freeze/hangs keyboard on stm32

macseem, 2024-07-15:
Pressing key assigned to a macros doesn't do anything
After that keyboard stops responding
Macros complexity doesn't matter, default one from documentation behaves the same
Key position/layer anything else doesn't change situation.

This is reproducing using on both github actions build and local west build
Both builds are successful and don't fail

Any other behaviour works as expected (hold-tap, mod-tap, etc), 

Board: STM32 f411ce black pill

Keymap configuration:
```C
#include <behaviors.dtsi>
#include <dt-bindings/zmk/keys.h>

/{
    macros {
        zed_em_kay: zed_em_kay {
            compatible = "zmk,behavior-macro";
            #binding-cells = <0>;
            bindings
                = <&macro_press &kp LSHFT>
                , <&macro_tap &kp Z &kp M &kp K>
                , <&macro_release &kp LSHFT>
                ;
        };
    };
    keymap {
        compatible = "zmk,keymap";
        default_layer {
            bindings = <
&zed_em_kay &none &none &none
&none       &none &none &none
&none       &none &none &none
            >;
            label = "Default layer";
        };
    };
};
```

Here is overlay
```C
/ {
    chosen {
      zmk,kscan = &default_kscan;
    };

    default_kscan: kscan {
      compatible = "zmk,kscan-gpio-matrix";
      label = "default_kscan";
      diode-direction = "col2row";

      col-gpios
        = <&blackpill 14 GPIO_ACTIVE_HIGH>
        , <&blackpill 15 GPIO_ACTIVE_HIGH>
        , <&blackpill 16 GPIO_ACTIVE_HIGH>
        , <&blackpill 17 GPIO_ACTIVE_HIGH>
        ;

      row-gpios 
        = <&blackpill 18 (GPIO_ACTIVE_HIGH | GPIO_PULL_DOWN)>
        , <&blackpill 19 (GPIO_ACTIVE_HIGH | GPIO_PULL_DOWN)>
        , <&blackpill 20 (GPIO_ACTIVE_HIGH | GPIO_PULL_DOWN)>
        ;

                
    };
};
```

caksoylar, 2024-07-15:
Can you also get logging output? https://zmk.dev/docs/development/usb-logging

macseem, 2024-07-15:
I tried to add logging, but it fails to compile, whenever I set flag on it just pops with lots of errors
like here https://github.com/macseem/zmk-config/actions/runs/9110620165

petejohanson, 2024-07-15:
> @caksoylar I tried to add logging, but it fails to compile, whenever I set flag on it just pops with lots of errors like here https://github.com/macseem/zmk-config/actions/runs/9110620165

See the updates docs on using the new logging snippet to enable this: https://zmk.dev/docs/development/usb-logging#usb-logging-snippet

I would *guess* that you're blowing the stack, but hard to tell for sure. You could try bumping the system work queue stack size too.

macseem, 2024-07-15:
Oh, haven't seen updated docs for usb logging, thanks for pointing, will try.
it defo feels like stack overflow or something similar. will try new logging and return with logs.

> You could try bumping the system work queue stack size too.
Not sure how to do this one, I don't have much experience with zephyr

petejohanson, 2024-07-15:
> Oh, haven't seen updated docs for usb logging, thanks for pointing, will try. it defo feels like stack overflow or something similar. will try new logging and return with logs.
>
>
> Not sure how to do [...]

You can set:

```
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048
```

For instance. If building locally, you can do a `west build -d build/some_build_dir -t menuconfig` to change it locally too (just don't build with -p/--pristine after or it will blow away config changes).

dhruvinsh, 2024-11-16:
Thanks @petejohanson  and @lesshonor 

I confirm that it is working for me, I was having the same issue. 

https://discord.com/channels/719497620560543766/1240770772352634970

https://github.com/dhruvinsh/zmk-config/commit/a0663c5884bd7008bc266362b0c54fd5345d10e3

ps. I have blackpill with tinyuf2.