zmkfirmware/zmk #2379: fix: `MACRO_PLACEHOLDER` causes `parse error: expected number or parenthesized expression`

jjangga0214, 2024-07-17:
Hi!

I have a config like this.

```
/ {
    macros {
        bt_to: bt_to {
            label = "bt_to";
            compatible = "zmk,behavior-macro-two-param";
            #binding-cells = <2>;  /* 2 params macro */
            bindings
                = <&macro_param_1to2>
                , <&bt BT_SEL MACRO_PLACEHOLDER>
                , <&macro_param_2to1>
                , <&to MACRO_PLACEHOLDER>;
        };
    };
};
```

And an error occurs on build. 

```sh
devicetree error: adv360.keymap:253 (column 26): parse error: expected number or parenthesized expression
```

It seems use of `MACRO_PLACEHOLDER` causes the error.

ReFil, 2024-07-17:
On an advantage 360 pro specific note, V2.0 config repositories are very outdated by now and as a result several features such as macros are not properly supported in it. There is a migration guide to the new v3.0 repositories available here: https://github.com/KinesisCorporation/Adv360-Pro-ZMK/blob/V3.0/UPGRADE.md

closed 2024-07-17 by caksoylar

jjangga0214, 2024-07-18:
Oh, thank you very much!