zmkfirmware/zmk #2382: fix: `&mo` in `&mo` doesn't seem to work

jjangga0214, 2024-07-18:
Hi!

I have a config like this.
(Details are omitted for brevity.) 

```
#define B 1
#define C 2

/ {
    keymap {
        compatible = "zmk,keymap"; 
        a { 
            &mo B  &kp A  &kp N0
        }
        b { 
            &trans &mo C  &kp N1
        }        
        c { 
            &trans &trans  &kp N2
        }
    }    
};
```

As we can see, the layer starts from `a`.
When I press (and keep holding) the first key, (obviously) the layer `b` is activated by `&mo B`.

However, I cannot activate the layer `c`. 
I expected holding the first and second key in sequence to activate the layer `c` by `&mo C`, but it doesn't.

caksoylar, 2024-07-18:
This works here, I assume you didn't try exactly the example. If not, please provide more details on your setup.

jjangga0214, 2024-07-19:
@caksoylar Hi! I am sorry! And thanks for the reply :)
You're right! It works.

My mistake was that I aligned layers by `a`, `c`, and `b`.

```
#define B 1
#define C 2

/ {
    keymap {
        compatible = "zmk,keymap"; 
        a { 
            &mo B  &kp A  &kp N0
        }
        c { 
            &trans &trans &kp N2
        }        
        b {
            &trans &mo C  &kp N1
        }        
    }    
};
```

, which should've been `a`, `b`, and `c`.

```
#define B 1
#define C 2

/ {
    keymap {
        compatible = "zmk,keymap"; 
        a { 
            &mo B  &kp A  &kp N0
        }
        b { 
            &trans &mo C  &kp N1
        }        
        c { 
            &trans &trans &kp N2
        }
    }    
};
```

I didn't know that the order of layers mattered.
I just thought `&mo` activates the target layer upon any other layers regardless of definition order.

I wonder if this behavior is documented because I haven't read this behavior.

caksoylar, 2024-07-20:
>  I just thought &mo activates the target layer upon any other layers regardless of definition order.

Yes, `&mo` always activates a layer however that doesn't mean the previous layers become inactive. If another layer that is currently active is defined later it will capture the key event first.

> I wonder if this behavior is documented because I haven't read this.

It is in theory documented in https://zmk.dev/docs/features/keymaps#layers but I am aware that it is too short and not clear, especially for newer users. #1284 is an open issue to improve that.

> And I'd like to ask whether this is an intended feature or a limitation.

I'd say it is intended, since it is needed to be able to use multiple active layers with transparent bindings. The alternative would be to have activation order determine priority, but I haven't seen that in any other firmware either.

closed 2024-07-20 by caksoylar