zmkfirmware/zmk #2385: Feature request: Only send the last pressed key

darix, 2024-07-21:
To basically implement what can be seen here

https://www.youtube.com/watch?v=Feny5bs2JCg

caksoylar, 2024-07-22:
I think this is related: https://github.com/qmk/qmk_firmware/pull/24000

Warning to readers: The code is GPL, see the [clean room policy](https://zmk.dev/docs/development/clean-room). I linked because the PR description/docs and naming can be useful.

darix, 2024-07-22:
yeah but 24000 only implements part of it. the resuming previous key part is not in this pull request. [see this comment and the reply](https://github.com/qmk/qmk_firmware/pull/24000#issuecomment-2241749830)

kovasap, 2024-07-25:
I have a suspicion this is already possible with zmk's existing behaviors, but I don't know how.  Either way, I'm also interested in this feature!