zmkfirmware/zmk #2395: Feature request: Passthrough BLE mouse to selected output

NilsIrl, 2024-07-30:
Effectively turn a zmk keyboard into a BLE KVM switch.

Currently when using a zmk keyboard with multiple BLE outputs when switching output for the keyboard other peripherals like mice obviously do not follow the output selection which means that multiple mice need to be used or a separate KVM switch. It would be cool to be able to connect a BLE mouse to a zmk keyboard and have the keyboard forward the events to the selected output.

Tips for someone who would want to have a crack at this:

> You'd need to look at creating a GATT client that scans for and connects to the pointer.
> Then subscribes to the right HID Over GATT (HoG) characteristics to recieve HID data, and the process/forward on our HID side.
> You can see our split central code for examples of scanning/GATT discovery, etc.

This module can relay input for another zmk device: https://github.com/badjeff/zmk-split-peripheral-input-relay

xudongzheng, 2024-07-31:
Related: #1732

ELLIOTTCABLE, 2025-03-02:
Although #1732 would definitely be cool (I have a few HID devices on my desk which I would love to 'follow' my keyboard's connection to either my Windows or Mac machines), it sounds more involved. This, however (bluetooth-only, acting as a forwarding host), would be _killer_. To not have to switch both my mouse and keyboard every time I switch machines? Further, to be able to have my Glove80's _four_ paired devices, instead of being limited by my Logitech MX Ergo's only-two-slots switching feature??

eynsai, 2026-01-13:
I definitely would find it useful if ZMK or a module could pair with consumer Bluetooth pointing devices and ingest incoming HID reports into ZMK's internal pointing device pipeline. I rely heavily on jointly processing keyboard and pointing device inputs, so just transparently passing the HID along untouched wouldn't be as useful to me. 

I use a Logitech MX Ergo trackball, and my current setup is absurd... The HID from the trackball goes into sekigon's keyboard quantizer, which converts it to QMK mouse reports, which I then convert to raw HID messages, which go to the host PC, which runs a host app that passes that raw HID to my keyboard, which then unpacks it into QMK mouse reports. Which is a ridiculous amount of complexity but it lets me manipulate the output of my trackball using my keyboard firmware. 

ZMK, with its support for multiple Bluetooth peripherals, seems well positioned to greatly simplify this mess I'm currently dealing with.