zmkfirmware/zmk #2400: Exception list for `require-prior-idle-ms`

alex35mil, 2024-08-03:
I configured home row mods using urob's approach, and it works great except for one case I keep bumping into. For example, when I hit ⌫ ⌫ ⌫ and then some character with mod, eg, ⌘ S or any shifted character, it often results in two characters printed instead of mod + character because of `require-prior-idle-ms`. Is it feasible to introduce a configuration param to exclude particular keys (like, ⌫ or even all non-alphabetic chars) from this rule?

imbryk, 2024-10-28:
+1 on this - I have the exact same use case for this exclude list
it could be defined as list of keys, just like `hold-trigger-key-positions`

alex35mil, 2025-08-02:
Not stale.

antoineco, 2025-11-06:
@rygwdn you seem to have come up with a working implementation, which I see you are [using](https://github.com/rygwdn/Adv360-Pro-ZMK/blob/7d24a7f3c9d58c29cc88c09adaaab08e50aa515f/config/west.yml#L14) inside your own firmware.
Have you considered submitting a PR to add that feature upstream?