zmkfirmware/zmk #2403: Toggle layer until an idle timeout has been reached

kovasap, 2024-08-04:
I'd like to try toggling a layer until a configurable number of seconds has elapsed, at which point the layer would toggle off.  Is there any way to do this in zmk currently?

caksoylar, 2024-08-04:
Not exactly as described. [Sticky layers](https://zmk.dev/docs/behaviors/sticky-layer) have a timeout but they'll also deactivate the layer before the timeout if any keycode is sent.

kovasap, 2024-08-04:
Right, I use those already!

caksoylar, 2024-08-04:
I think a clever application of the tri-state behavior (#1366, also available as a [module](https://github.com/dhruvinsh/zmk-tri-state)) using the `timeout-ms` property and `&tog` bindings might work for this. But you might need to set `ignored-*` properties as well.