zmkfirmware/zmk #2408: Synced sleep bewteen central/peripherals

Genteure, 2024-08-07:
Would be nice to have an option to sync sleep timers.
Currently key presses on a peripheral keeps the central awake but not the reverse. I imagine with a multi-peripheral setup (e.g. with a dongle) key presses on peripheral A would also not keep peripheral B awake.
This leads to the peripheral side occasionally sleeping on it's own. Although wake-up and reconnect is pretty quick in my experience, there's still a few seconds of delay.

Nick-Munnich, 2024-08-07:
Personally I'd say your suggested function should be how it behaves, I don't see the reason why you would want to have the option to *not* sync sleep timers.

angweekiat, 2024-09-05:
Hi @Genteure , I took a stab at this, and shared my changes above, based on syncing the last activity timing, rather than sleep timers. One issue I realized is that in deep sleep mode, the BLE capability is turned off, so it's not possible to wake either device from either side (to my knowledge), but the changes do prevent the peripheral from going into deep sleep, if the central device is used, and the timing synced to the peripheral

angweekiat, 2024-09-06:
Hmm @Genteure , I was reading through some other issues when a realisation occurred, the last activity timers are synced, but the state is not. If the central is powered by USB, it'll never go to sleep, but the last activity time is still synced, and can result in the peripheral going to idle mode, while the central is in active state.

Some users may be ok with this (saves battery on the peripheral), while some may prefer the alternative where even the states are synchronised. Maybe it would be better to explicitly sync the states rather than indirectly through the last activity timers.

Just wanted to check if this use case crossed your mind when raising the issue, and your thoughts on it?

Genteure, 2024-09-06:
I don't really have a strong opinion but to me syncing time numbers feels better than completely synchronized sleep state.

I can think of four scenarios where the central is connected to USB but the peripherals are not:

1. Using as a USB keyboard w/ wireless split.
2. Dongle mode.
3. Simply charging the battery.
4. Wired split (when zmk fully supports it)

For scenario 4, users would probably not enable sleep to begin with.
For the other three, the peripheral(s) should still able to sleep when all parts are not actively used, otherwise it defeats the whole purpose of having a sleep feature isn't it. If a user doesn't want to have their keyboard go to sleep they can just set `CONFIG_ZMK_SLEEP=n`.

Also a peripheral should still be able to sleep on it's own to save battery when it's not connected to the central.

aeros281, 2025-06-11:
I have a gaming use case where I use a set of layers focused on the central half of the keyboard, since my right hand is obviously operating the mouse.

To minimize latency, I connect the central half via USB. The peripheral half isn't connected to anything, as I assume it will communicate wirelessly with the central half even if I connect it via USB.

However, after some time, the peripheral goes to sleep on its own, which I don’t want — sometimes I switch back to non-gaming layers to chat or do other tasks, and it’s inconvenient to have to wake up the peripheral.

Having a setting to synchronize idle and sleep times between the two halves would be really helpful.