zmkfirmware/zmk #2410: Feature request: Hardware troubleshooting shield

Nick-Munnich, 2024-08-07:
For hardware troubleshooting, it would be convenient to have a shield in a similar vein to `settings_reset` which behaves as follows:
1. Configures all GPIO pins as input with a high pullup
2. Whenever any GPIO pin is pulled low (connected to ground), set all other GPIO pins to low while the triggered GPIO pin is low

Because some boards have shorts between GPIO pins and reset, there needs to be a mechanism in devicetree to exclude some GPIO pins from the above behaviour, rather leaving them hanging.

caksoylar, 2024-08-13:
Maybe we can start with a simpler version without 2.? e.g. a Pro Micro version with direct pins mapping to a macro each that prints the pro micro pin number. And another for the Xiao interconnect perhaps.

Nick-Munnich, 2024-08-13:
That would probably be a better approach entirely - somehow the possibility of using the usb connection completely slipped in my mind.

PR #2360 (2024-10-31): docs: Add a hardware issues page under troubleshooting
This PR adds a "hardware issues" page under troubleshooting. It discusses:
- Identifying problems with pins
- Firmware changes needed after soldering a bodge wire
- Common physical problems with antennas
- Advice for troubleshooting the physical problems

In the process, there are also two new shields to help with troubleshooting: `tester_pro_micro` and `tester_xiao`. This partially resolves #2410, though I would really like a more general solution eventually.
closed 2024-10-31 by caksoylar