zmkfirmware/zmk #2417: [GH Actions] Make it possible to run build process from subdirectory

suttonbradley, 2024-08-13:
I'm hoping to be able to contain my keyboard design (ergogen/other) files and my ZMK config within the same directory in a clean way, like having a subdirectory `zmk-config` and having the github actions run with that dir as a working directory.

I see that the GH actions yaml takes [four inputs](https://github.com/zmkfirmware/zmk/blob/fb3af5327e7e42b826c9fffc2cd26c60c7ec8c9e/.github/workflows/build-user-config.yml#L5), and I tried [setting those](https://github.com/suttonbradley/aldrin/blob/e0ffa06e936d283a74e02b057f92e5806f2cfeb6/.github/workflows/build.yml#L7-L8) in my usage of the `build-user-config.yaml` template.

Links for reference:
- [Discrete repo](https://github.com/suttonbradley/aldrin-zmk) (GH actions works)
- [Combo repo](https://github.com/suttonbradley/aldrin/tree/zmk) (zmk branch, as linked)
- [Failing GH actions on combo repo](https://github.com/suttonbradley/aldrin/actions/runs/10367234503/job/28698220177)

Nick-Munnich, 2024-08-13:
I think you should be able to do this by [setting the working directory](https://docs.github.com/en/actions/writing-workflows/workflow-syntax-for-github-actions#defaultsrun). That said I'd push back against documenting this, I don't think this approach should be recommended.

kaievns, 2024-10-10:
i just run into the same issue, and I'm not sure why this is contested. seems pretty reasonable request.

yes submodules might be an option but they're also a mess to maintain, especially if on is developing several keyboards. adding a couple of lines to a config is a better experience however you turn it

@suttonbradley did you by any chance figure out how to make the default working directory setting work?