zmkfirmware/zmk #2426: Request: space bar at consumer level

sirus713, 2024-08-15:
When SPACE is pressed it sends unicode 231  hex 0xe7

I need it to send unicode 32 hex 0x20.

Microsoft only sees VK_PACKET

petejohanson, 2024-08-15:
So, ZMK is for HID devices, so we would need some research/understanding of what HID *usage* you're wanting to use that will trigger Windows to trigger that key code at the OS level

sirus713, 2024-08-15:
Ideally I would like to use the playback button in some of the 3d creative softwares. When the space key is pressed there is no action in UI of the software I'm using. Because the only code it is receiving happens to match the rest of alpha-numeric codes that get sent. They all come through as VK_PACKET on unicode 231

The keystrokes act as they should, they type just fine, and maybe that's why the space bar won't behave the way I need it to when an active window has a dedicated hotkey....it thinks I'm typing and not activating an action

Not sure. 

Hope this explains more