zmkfirmware/zmk #2434: Feature Request: Bluetooth Toggle Hotkey

JackDV, 2024-08-21:
Currently, when my ZMK PCB is connected to a profile, and I hit `BT_DISC` to disconnect to the profile, the keyboard will automatically repair to that profile if nothing else is available. 

What I am looking for is a way to disable bluetooth pairing, meaning my PCB can remain turned on, and not connected to a bluetooth profile, even if one is pairable.

### Example situation
My PCB has a single profile paired to it, my keyboard is turned on and paired to that profile.

This is what currently happens
1. I hit `BT_DISC` to disconnect from the profile
2. My keyboard disconnects from the profile
3. My keyboard automatically reconnects from the profile

This is what I would like to happen

1. I hit `BT_DISC` to disconnect from the profile
2. My keyboard disconnects from the profile
3. My keyboard stays disconnected from the profile

When I brought this up as a question in the ZMK discord, it was suggested a `BT_OFF` key be added to the `&bt` list of commands, which seems like it would turn off bluetooth pairing on the keyboard side, meaning there could be pairable profiles available and the keyboard would remain unpaired.

caksoylar, 2024-08-21:
Related: #1998

Nick-Munnich, 2024-10-16:
#2546 should resolve this