zmkfirmware/zmk #2435: Add file markings to code blocks in the docs

Nick-Munnich, 2024-08-21:
We have a lot of code blocks in the docs. It would be useful if someone could go through and mark the blocks with `title=` where it could be helpful for readers to understand where the code blocks belong. Could be a useful task for someone who wants to become more familiar with the way things work.

nmunnich, 2025-09-09:
That's a fair point - for end-users it may indeed not be helpful. If you've already done the task, could you perhaps open a draft PR so that we can look at the end result? I think you're probably right in that the general configuration page along with everything in Development would be the places where the titles would be most useful.

nmunnich, 2025-12-26:
I think looking at the docs now we have sufficient titles indeed.

closed 2025-12-26 by nmunnich