zmkfirmware/zmk #2461: As of mid-day 9/6, new GitHub builds cause BT signing error for intra-keyboard handshaking

lazerwalker, 2024-09-07:
I have a wireless Corne using two nice!nanos and two nice!views.

Starting earlier today, new builds seem to error out when trying to connect both halves.

The way I experience this: if the right (secondary) half is off, the left (primary) half works fine. As soon as I turn the right half on, the left half stops working, and the display (a nice!view) freezes. The `<bt_att> Error signing data` error can be seen below from a USB debug log.

I am building my firmware using the default GitHub Actions setup — in case it's helpful, the repo running the workflows is https://github.com/lazerwalker/zmk-typeractive-corne-noview.

I've confirmed this issue happens with any build I've made since about mid-day (PST) 9/6. If I flash a firmware generated prior than that, it works as expected. If I revert the git repo to a SHA that had previously build successfully, the newly-generated firmware does not work.

To determine whether the issue was caused by a change to `zmkfirmware/zmk/.github/workflows/build-user-config.yml` (i.e. the workflow file loaded by my useland workflow), I pegged it to the SHA of a known good build instead of the default target of `main`. That did not fix the issue.

I've encountered at least one other person (on the Typeractive discord) who has started experiencing this issue as well.

A gut suspicion I haven't investigated yet is whether a new stable build has been pushed to docker in the last day. Curious if anyone else has run into this or has any suspicions / tips to debug.

```
[00:00:35.574,096] <dbg> zmk: split_central_eir_found: Found the split service
[00:00:35.574,096] <dbg> zmk: zmk_ble_put_peripheral_addr: Found existing peripheral address in slot 0
[00:00:35.574,127] <dbg> zmk: stop_scanning: Stopping peripheral scanning
[00:00:35.574,371] <dbg> zmk: split_central_eir_found: Initiating new connection
[00:00:35.681,854] <dbg> zmk: connected: Connected thread: 0x20006ce8
[00:00:35.681,854] <dbg> zmk: connected: SKIPPING FOR ROLE 0
[00:00:35.681,976] <dbg> zmk: split_central_connected: Connected: D8:0C:A0:1C:96:40 (random)
[00:00:35.681,976] <dbg> zmk: split_central_process_connection: Current security for connection: 1
[00:00:35.682,037] <dbg> zmk: split_central_process_connection: New connection params: Interval: 6, Latency: 30, PHY: 1
[00:00:35.682,037] <dbg> zmk: start_scanning: All devices are connected, scanning is unnecessary
[00:00:35.691,528] <dbg> zmk: split_central_service_discovery_func: [ATTRIBUTE] handle 28
[00:00:35.691,558] <dbg> zmk: split_central_service_discovery_func: Found split service
[00:00:35.705,810] <dbg> zmk: split_central_chrc_discovery_func: [ATTRIBUTE] handle 2
[00:00:35.705,841] <dbg> zmk: split_central_chrc_discovery_func: [ATTRIBUTE] handle 5
[00:00:35.736,236] <dbg> zmk: split_central_chrc_discovery_func: [ATTRIBUTE] handle 7
[00:00:35.736,267] <dbg> zmk: split_central_chrc_discovery_func: [ATTRIBUTE] handle 10
[00:00:35.750,793] <dbg> zmk: split_central_chrc_discovery_func: [ATTRIBUTE] handle 12
[00:00:35.750,823] <dbg> zmk: split_central_chrc_discovery_func: [ATTRIBUTE] handle 14
[00:00:35.773,284] <dbg> zmk: split_central_chrc_discovery_func: [ATTRIBUTE] handle 17
[00:00:35.773,315] <dbg> zmk: split_central_chrc_discovery_func: [ATTRIBUTE] handle 22
[00:00:35.788,269] <dbg> zmk: split_central_chrc_discovery_func: [ATTRIBUTE] handle 24
[00:00:35.788,330] <dbg> zmk: split_central_chrc_discovery_func: [ATTRIBUTE] handle 26
[00:00:35.803,344] <dbg> zmk: split_central_chrc_discovery_func: [ATTRIBUTE] handle 29
[00:00:35.803,344] <dbg> zmk: split_central_chrc_discovery_func: Found position state characteristic
[00:00:35.803,436] <dbg> zmk: split_central_subscribe: [SUBSCRIBED]
[00:00:35.833,343] <dbg> zmk: split_central_chrc_discovery_func: [ATTRIBUTE] handle 32
[00:00:35.833,374] <dbg> zmk: split_central_chrc_discovery_func: Found run behavior handle
[00:00:35.863,342] <dbg> zmk: split_central_chrc_discovery_func: [ATTRIBUTE] handle 35
[00:00:35.863,372] <dbg> zmk: split_central_chrc_discovery_func: Found select physical layout handle
[00:00:35.863,433] <err> bt_att: Error signing data
```

petejohanson, 2024-09-07:
I did merge a split related change earlier today. Is it improved if you *also* flash an updated image to the peripheral/right half? I may have flubbed the backwards compat for the central code.

lazerwalker, 2024-09-07:
It still happens if both halves are running 'matching' firmware (generated from the same Actions build), yeah.

petejohanson, 2024-09-07:
Ok, I'll revert the PR for now and investigate tomorrow

petejohanson, 2024-09-07:
Please test updated `main` and report back if that resolved this for you. Thanks!

lazerwalker, 2024-09-07:
I'm headed to bed, but will text first thing in the morning. Appreciate the quick revert!!

I also dropped this issue link into a Discord thread with a few other folks who were having issues, it's possible one of them may be able to report back before me.

theesue, 2024-09-07:
I just tested a build that did not work since this morning and it is working now. Thanks!

narvcharv, 2024-09-07:
I was seeing the exact same issue yesterday - revert works for me as well

lazerwalker, 2024-09-07:
Working for me as well now! Closing this out.

closed 2024-09-07 by lazerwalker