zmkfirmware/zmk #2467: CONFIG_ZMK_WIDGET_LAYER_STATUS not working

quiquos, 2024-09-08:
A few days ago, CONFIG_ZMK_WIDGET_LAYER_STATUS stopped showing the layer we are in.

this is my conf file:
```
CONFIG_ZMK_KEYBOARD_NAME="Quiquos Corne"
CONFIG_ZMK_DISPLAY=y
CONFIG_ZMK_WIDGET_LAYER_STATUS=y
CONFIG_ZMK_WIDGET_BATTERY_STATUS=y
CONFIG_ZMK_WIDGET_BATTERY_STATUS_SHOW_PERCENTAGE=y
CONFIG_ZMK_WIDGET_OUTPUT_STATUS=y
CONFIG_ZMK_WIDGET_WPM_STATUS=y
CONFIG_ZMK_SLEEP=y
CONFIG_ZMK_IDLE_SLEEP_TIMEOUT=900000
CONFIG_BT_CTLR_TX_PWR_PLUS_8=y
CONFIG_ZMK_PM_SOFT_OFF=y
```

If I flash a version from, for example, a month ago, it works great.
The same code compiled last week won't show the layer number we are in.

I tried something I read that CONFIG_ZMK_WIDGET... are no longer needed to put in conf file, but even so, it doesn't work.

My keyboard is the Split Corne wireless connected via BT (also with cable, layer is not displayed)

caksoylar, 2024-09-09:
Does it work if you add `display-name` properties to the layer nodes? e.g. `display-name = "Base";`

caksoylar, 2024-09-09:
No, it is an additional property that goes in the node like I mentioned above. See examples in https://github.com/zmkfirmware/zmk/blob/1c48f64730a479115ae827c342c1063b2b8e93ca/app/boards/shields/reviung34/reviung34.keymap#L28

quiquos, 2024-09-09:
Nice, it worked.

Actually, it's even better, because I thought that I could only display a number as a layer "name" to be displayed, and now I have the name I want.

Thanks

Grazfather, 2024-09-19:
I added this to the docs in #2460

petejohanson, 2024-09-19:
Flease give https://github.com/zmkfirmware/zmk/pull/2491 a test.

caksoylar, 2024-09-23:
Closing given #2491 is merged, please re-open if still an issue.

closed 2024-09-23 by caksoylar