zmkfirmware/zmk #2471: build custom shield error : parse error: expected number or parenthesized expression

exqmjmz, 2024-09-09:
I want to customize a split keyboard for corne70
I copied corne's shield file and modified it
my custom files

[corne70.zip](https://github.com/user-attachments/files/16932999/corne70.zip)

But when I built my own custom shield, an error occurred

build command:
```
 west build -p -b nice_nano -- -DSHIELD=corne_left
```

```
-- Found devicetree overlay: /workspaces/zmk/app/boards/shields/corne70/corne_left.overlay
-- Found devicetree overlay: /workspaces/zmk/app/boards/shields/corne70/boards/nice_nano.overlay
-- Found devicetree overlay: /workspaces/zmk/app/boards/shields/corne70/corne.keymap
devicetree error: /workspaces/zmk/app/boards/shields/corne70/corne.keymap:22 (column 105): parse error: expected number or parenthesized expression
-- In: /workspaces/zmk/app/build/zephyr, command: /usr/bin/python3;/workspaces/zmk/zephyr/scripts/dts/gen_defines.py;--dts;/workspaces/zmk/app/build/zephyr/zephyr.dts.pre;--dtc-flags;'-Wno-unique_unit_address_if_enabled';--bindings-dirs;/workspaces/zmk/app/module/dts/bindings;/workspaces/zmk/app/dts/bindings;/workspaces/zmk/zephyr/dts/bindings;--header-out;/workspaces/zmk/app/build/zephyr/include/generated/devicetree_generated.h.new;--dts-out;/workspaces/zmk/app/build/zephyr/zephyr.dts.new;--edt-pickle-out;/workspaces/zmk/app/build/zephyr/edt.pickle;--vendor-prefixes;/workspaces/zmk/app/dts/bindings/vendor-prefixes.txt;--vendor-prefixes;/workspaces/zmk/zephyr/dts/bindings/vendor-prefixes.txt
CMake Error at /workspaces/zmk/zephyr/cmake/modules/dts.cmake:279 (message):
  gen_defines.py failed with return code: 1
Call Stack (most recent call first):
  /workspaces/zmk/zephyr/cmake/modules/zephyr_default.cmake:129 (include)
  /workspaces/zmk/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /workspaces/zmk/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:8 (find_package)
```

my keymap
```
/*
 * Copyright (c) 2020 The ZMK Contributors
 *
 * SPDX-License-Identifier: MIT
 */

#include <behaviors.dtsi>
#include <dt-bindings/zmk/keys.h>
#include <dt-bindings/zmk/bt.h>

/ {
        keymap {
                compatible = "zmk,keymap";

                default_layer {
                        bindings = <
                        &kp N1	&kp N2	&kp N3	&kp N4	&kp N5	&kp LPAR	&kp RPAR	&kp N6	&kp N7	&kp N8	&kp N9	&kp N0
                        &kp Q	&kp W	&kp E	&kp R	&kp T	&kp DEL	&kp TILDE2	&kp Y	&kp U	&kp I	&kp O	&kp P
                        &kp A	&kp S	&kp D	&kp F	&kp G	&kp DLLR	&kp EQUAL	&kp H	&kp J	&kp K	&kp L	&kp SEMI
                        &kp TAB	&kp Z	&kp X	&kp C	&kp V	&kp B	&kp N	&kp M	&kp BSLH	&kp FSLH	&kp UP	&kp BSPC
                        &kp CAPS	&kp LSHIFT	&kp LCTRL	&kp LALT	&kp LCMD	&none	&none	&kp RSHIFT	&kp LEFT	&kp DOWN	&kp RIGHT	&kp ENTER
                        &kp SPACE	&kp RBKT	&kp LBKT	&kp EXCL	&kp MINUS	&none	&none	&kp FSLH	&kp SQT	&kp CMMA	&kp DOT	&kp SPACE
                        >;
                };
        };
};

```

But when I remove the last row, it compiles fine
How should this problem be solved?

Nick-Munnich, 2024-09-09:
Please read through [our troubleshooting pages](https://zmk.dev/docs/troubleshooting/building-issues) for errors such as these; I believe that this error should be covered under them.

exqmjmz, 2024-09-09:
[image]
Each row has the same number of columns

exqmjmz, 2024-09-09:
> The troubleshooting page would have pointed you to that exact line, stating that you made a mistake when defining your keymap for that line. For this case specifically, `CMMA` is not a [keycode](https://zmk.dev/docs/keymaps/list-of-keycodes) - there may be other errors, but that is what I spotted.

[image]
You can find it here
dt-bindings/zmk/keys.h

😂The problem is strange to solve
I replicated the last line, then deleted the row and pasted it back.
It is normal to compile again.

closed 2024-09-09 by exqmjmz