zmkfirmware/zmk #2489: `zmk,matrix-transform` invalidates firmware build

DanielAtCosmicDNA, 2024-09-18:
As can be seen on https://github.com/typeractivexyz/corne-wireless-5-col-view-zmk-config/issues/6. The option `zmk,matrix-transform` of https://zmk.dev/docs/config/kscan#devicetree invalidates the firmware build with the following build error:

>    17 | #error "To use physical layouts, remove the chosen `zmk,matrix-transform` value."
>       |  ^~~~~

petejohanson, 2024-09-18:
@DanielAtCosmicDNA Can you please give #2490 a try?

DanielAtCosmicDNA, 2024-09-20:
> @DanielAtCosmicDNA Can you please give #2490 a try?

Sure. I have just retriggered [the build](https://github.com/CosmicDNA/corne-wireless-5-col-view-zmk-config/actions/runs/10813360993) and now it is generating the firmware.

closed 2024-09-20 by DanielAtCosmicDNA