zmkfirmware/zmk #2493: Sideband behaviour only allows for any bindings of single word

Trap101, 2024-09-22:
It seems that `kscan-sideband-behaviors` only allows for bindings in the form of `&capsword` or `&ble` etc. and does not allow for anyone like `&kp A` as there is a space in the middle. 
So this does not work and returns the error `parse error: expected number or parenthesized expression`
```
    endpoint_sideband_behaviour {
        compatible = "zmk,kscan-sideband-behaviors";
        auto-enable;
        kscan = <&kscan_sp3t_toggle>; 
        first {
            column = <0>;
            bindings = <&kp F>;
        };
        second {
            column = <1>;
            bindings = <&kp J>;
        };
    };
```
while this works 
```
    endpoint_sideband_behaviour {
        compatible = "zmk,kscan-sideband-behaviors";
        auto-enable;
        kscan = <&kscan_sp3t_toggle>; 
        first {
            column = <0>;
            bindings = <&caps_word>;
        };
        second {
            column = <1>;
            bindings = <&caps_word>;
        };
    };
```
here is my repo which should contain a good history of failing and succeeding in github action forms
https://github.com/Trap101/my-dactyl-keyboard

petejohanson, 2024-09-22:
You're probably receiving that error because you're not `#include`-ing the keys.h header to bring in the defines for `A`, `B`, etc.

nmunnich, 2025-12-28:
Closing, as an answer was given and no further comments have been made.

closed 2025-12-28 by nmunnich