zmkfirmware/zmk #2511: Update new behavior example code

joelspadin, 2024-09-27:
Some of the example code at https://zmk.dev/docs/development/new-behavior#creating-the-driver-c is either outdated or incomplete. The following issues should be fixed:

1. `#if DT_HAS_COMPAT_STATUS_OKAY(DT_DRV_COMPAT)` in the C file should be replaced with
    ```kconfig
    config ZMK_BEHAVIOR_<NAME>
        bool
        default y
        depends on DT_HAS_ZMK_BEHAVIOR_<NAME>_ENABLED
    ```
    in Kconfig.behaviors and
    ```cmake
    target_sources_ifdef(CONFIG_ZMK_BEHAVIOR_<NAME> app PRIVATE src/behaviors/<file>.c)
    ```
    in CMakeLists.txt
2. The code example with `BEHAVIOR_DT_INST_DEFINE(0, ...` shows how to create a behavior that only supports one instance. It includes a comment that implies that simply switching `0` to `n` will add support for multiple instances, but this is incorrect. The correct structure for such a behavior is
    ```c
    #define MY_BEHAVIOR_INST(n) \
        static const struct my_behavior_config my_behavior_config_##n = { ... }; \
        static struct my_behavior_data my_behavior_data_##n = { ... }; \
        BEHAVIOR_DT_INST_DEFINE(n, my_behavior_init, NULL, &my_behavior_data_##n, &my_behavior_config_##n, \
                                POST_KERNEL, CONFIG_KERNEL_INIT_PRIORITY_DEFAULT, &my_behavior_driver_api);
    
    DT_INST_FOREACH_STATUS_OKAY(MY_BEHAVIOR_INST)
    ```
    We could either make this the only example, then describe how to simplify it if your behavior only supports one instance, or we could have examples of both formats (though having two examples could be more confusing and harder to maintain).

caksoylar, 2024-11-27:
One more: document using `zmk_behavior_invoke_binding` instead of `behavior_keymap_binding_pressed` and `behavior_keymap_binding_released`.

caksoylar, 2025-01-13:
I think we should consider documenting creating this as a module by default, as well.

closed 2025-11-17 by caksoylar